SlashDevin/NeoGPS #1: Compilation error using Arduino Due or Maple Mini

SensorsIot, 2015-09-09:
I get compilation errors if I try to compile the NMEAtest script on a 32 bit platform (with IDE1.6.5).

The first error is:

In file included from C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS/NMEAGPS.h:25:0,
                 from NMEAtest.ino:21:
C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS/NMEAGPS_cfg.h:155:2: error: #error You must define NMEAGPS_DERIVED_TYPES in NMEAGPS.h in order to parse Talker and/or Mfr IDs
 #error You must define NMEAGPS_DERIVED_TYPES in NMEAGPS.h in order to parse Talker and/or Mfr IDs
  ^
Error compiling.
-----------------------------------------
If I comment this one, I get the next error: 

C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseLat(char)':
C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS\NMEAGPS.cpp:807:32: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lat' to 'int32_t& {aka long int&}'
     parseDDDMM( m_fix.lat, chr );
                                ^
C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseLon(char)':
C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS\NMEAGPS.cpp:828:32: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lon' to 'int32_t& {aka long int&}'
     parseDDDMM( m_fix.lon, chr );
                                ^
C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseHDOP(char)':
C:\Users\Andreas\Dropbox\ASP\Arduino\Sketches\libraries\NeoGPS\NMEAGPS.cpp:897:38: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::hdop' to 'uint16_t& {aka short unsigned int&}'
   if (parseFloat( m_fix.hdop, chr, 3 ))
                                      ^
Error compiling.

I would like to use your great library on a 32 bit platform because of speed, double precison, and many serial connections.

SlashDevin, 2015-09-13:
PARSE_MFR_ID and NMEAGPS_DERIVED_TYPES in NMEAGPS_cfg.h should have been commented out for the default configuration.  My apologies.

For the "cannot bind packed field" errors, please comment out NEOGPS_PACKED_DATA in NeoGPS_cfg.h.  This was not described in [Configurations](/SlashDevin/NeoGPS/blob/master/doc/Configurations.md), apologies again.  I added this option for another user that was trying NeoGPS on a 32-bit platform, so I hope this is all you need.

closed 2015-09-13 by SlashDevin

SensorsIot, 2015-09-14:
If I do your suggested changes, the compilation runs through for a Maple Mini. 
So far, I was not able to test the working of the example with a real GPS module.

Thank you for your help!

commit 429d6a4f (2015-09-13): Fix #1, invalid default configuration
commit 4ce6b04b (2015-09-14): Propagate fix #1 to example configs