SlashDevin/NeoGPS #3: DUE version

CANhaozhe, 2016-06-15:
Hi, Devin,
Current version is not working in my DUE.
Do you have Auduino/DUE version for this NeoGPS?
Thanks

SlashDevin, 2016-06-17:
More details, please.

CANhaozhe, 2016-06-24:
Hi, Devin,
Thanks for reply.

In Arduino IDE (version 1.6.9):
1. If I choose the board of Arduino Yun the compiling sketch is successfully as following.
>>Sketch uses 10,950 bytes (38%) of program storage space. Maximum is 28,672 bytes.
>>Global variables use 453 bytes (17%) of dynamic memory, leaving 2,107 bytes for local variables. >>Maximum is 2,560 bytes.

2. If I choose the board of Arduino DUE (program port) the compiling sketch is not passed with following  errors.
>>In file included from ...\NMEA_2\NMEA\NMEA.ino:2:0: sketch\NMEAGPS.h: In member >>function 'void NMEAGPS::lock() const':
>>NMEAGPS.h:317: error: 'cli' was not declared in this scope
>>      cli();
>>          ^
>>sketch\NMEAGPS.h: In member function 'void NMEAGPS::unlock() const':
>>NMEAGPS.h:323: error: 'sei' was not declared in this scope
>>           sei();
>>               ^
>>exit status 1
>>'cli' was not declared in this scope

Thanks

CANhaozhe, 2016-06-24:
2. If I choose the board of Arduino DUE (program port) the compiling sketch is not passed with following errors:
..\NMEA_2\NMEA\NMEA.ino:2:0:
sketch\NMEAGPS.h: In member function 'void NMEAGPS::lock() const':
NMEAGPS.h:317: error: 'cli' was not declared in this scope
           cli();
               ^
sketch\NMEAGPS.h: In member function 'void NMEAGPS::unlock() const':
NMEAGPS.h:323: error: 'sei' was not declared in this scope
           sei();
               ^
exit status 1
'cli' was not declared in this scope

SlashDevin, 2016-06-24:
Is NMEAGPS_INTERRUPT_PROCESSING enabled?  What other config items are different from the default?

For a quick fix, try replacing `cli()` with `noInterrupts()` and `sei()` with `interrupts()`.

CANhaozhe, 2016-07-01:
After replacing  cli()  with  noInterrupts()  and  sei()  with  interrupts().

In file included from NMEAGPS.cpp:19:0:
NMEAGPS.h: In member function 'void NMEAGPS::lock() const':
NMEAGPS.h:317: error: 'noInterrupts' was not declared in this scope
           noInterrupts();
                        ^
NMEAGPS.h: In member function 'void NMEAGPS::unlock() const':
NMEAGPS.h:323: error: 'interrupts' was not declared in this scope
           interrupts();
                      ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parseGSV(char)':
NMEAGPS.cpp:775: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::satellites[((int)((NMEAGPS*)this)->NMEAGPS::sat_count)].NMEAGPS::satellite_view_t::azimuth' to 'uint16_t& {aka short unsigned int&}'
                       parseInt( satellites[sat_count].azimuth, chr );
                                                                    ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parseLat(char)':
NMEAGPS.cpp:1244: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lat' to 'int32_t& {aka long int&}'
         );
         ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parseLon(char)':
NMEAGPS.cpp:1283: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lon' to 'int32_t& {aka long int&}'
         );
         ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parseHDOP(char)':
NMEAGPS.cpp:1392: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::hdop' to 'uint16_t& {aka short unsigned int&}'
     if (parseFloat( m_fix.hdop, chr, 3 ))
                                        ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parseVDOP(char)':
NMEAGPS.cpp:1407: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::vdop' to 'uint16_t& {aka short unsigned int&}'
     if (parseFloat( m_fix.vdop, chr, 3 ))
                                        ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parsePDOP(char)':
NMEAGPS.cpp:1422: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::pdop' to 'uint16_t& {aka short unsigned int&}'
     if (parseFloat( m_fix.pdop, chr, 3 ))
                                        ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parse_lat_err(char)':
NMEAGPS.cpp:1437: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lat_err_cm' to 'uint16_t& {aka short unsigned int&}'
     if (parseFloat( m_fix.lat_err_cm, chr, 2 ))
                                              ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parse_lon_err(char)':
NMEAGPS.cpp:1452: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lon_err_cm' to 'uint16_t& {aka short unsigned int&}'
     if (parseFloat( m_fix.lon_err_cm, chr, 2 ))
                                              ^
NMEAGPS.cpp: In member function 'bool NMEAGPS::parse_alt_err(char)':
NMEAGPS.cpp:1467: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::alt_err_cm' to 'uint16_t& {aka short unsigned int&}'
     if (parseFloat( m_fix.alt_err_cm, chr, 2 ))
                                              ^
'noInterrupts' was not declared in this scope

thanks anyway

commit 9f322c8c (2016-08-19): Fix #3 DUE version

* Use DUE Serial1 for the gps_port by default
* Pack structures only one AVR, not DUE
closed 2016-08-19 by SlashDevin