SlashDevin/NeoGPS #5: Using NeoGPS to store RAW Data

CesMak, 2016-07-15:
Hey,

I have another questions. I simply want to connect my uBlox GPS serial port to my adruino uno and read in NMEA sentences as well as RAW messages(RAWX, SFRBX,MEASX) and then store them to a microSD card (with openLog).

It works fine so far for storing NMEA sentences. However if I want to store RAW messages my code has some problems..... 
Thats why I want to ask how can be determined the ("end of line for raw messages?") or how can I use NEOGPS to simply print all the incoming data from the serial port of my gps module. 

My code so far was: 
```
String inData;

void setup() {
    Serial.begin(38400);
}

void loop() {
    while (Serial.available() > 0)
    {
        char recieved = Serial.read();
        inData += recieved; 

        // Process message when new line character is recieved
        if (recieved == '\n')
        {
            //Serial.print("Arduino Received: ");
            Serial.print(inData);

            inData = ""; // Clear recieved buffer
        }
    }

}
```
I mean if I connect my Serial Port of my gps module with a serial to usb converter and have a look at the data with RealTerm then all the linebrakes and so on is managed fine.

My goal is to store NMEA data and RAW data on an microSD card so that this file can afterwards be opened by ublox and when playing it it will recognice the RAW messages as well. 

Everything with serial.print will be written to my sd card.

Thanks! :)

SlashDevin, 2016-07-15:
> My goal is to store NMEA data and RAW data on an microSD card so that this file can afterwards be opened by <s>ublox</s> **u-center**
You do mean u-center, correct?  Ublox is the manufacturer.

NeoGPS is really for parsing the raw data into its `fix` structure members.  You can do both in one sketch if you want: you can write all the characters to the SD file **and** pass them to NeoGPS for parsing.  Do you need to use the field values (e.g., lat/lon or speed) in your sketch?

CesMak, 2016-07-15:
yeah ups I meant u-Center sry. 

Yeah I just want to write all the characters to my microSD card. I do not need any parsing just storing the data correctely. That is why i do not need any field values in my sketch.

My question just is how to use NEOGPS to just read in all the characters (also Raw data) correctely and write them to my microSD card?

Would be glad if you could give me a short example how to do that :)

Thanks :)

SlashDevin, 2016-07-15:
> My question just is how to use NEOGPS to just read in all the characters (also Raw data) correctely and write them to my microSD card?

You don't.  Just read characters from the serial port and write them to the SD file:

    // Pick one for the GPS device, in order of preference:
    //
    // 1)  Mega *ONLY*  (RX1 to GPS TX, TX1 to GPS RX).  Could also use Serial2 or Serial3
    //HardwareSerial & gps_port = Serial1;

    // 2) UNO pins 8 (RX to GPS TX) & 9 (TX to GPS RX)
    //#include <AltSoftSerial.h>
    //AltSoftSerial gps_port; 

    // 3) UNO, any two pin numbers.
    //#include <NeoSWSerial.h>
    //NeoSWSerial gps_port( 2, 3 ); // connected to GPS Tx,Rx
    
    void setup()
    {
      gps_port.begin( 9600 ); // for reading GPS data
      Serial.begin( 9600 ); // for printing to Serial monitor
     ...open SD file here...
    }
    
    void loop()
    {
      if (gps_port.available())
        sdFile.write( gps_port.read() ); // read a char and write it to the file
    }

NeoGPS is not needed.  You don't even need to watch for the newline.  Just read a char and write a char.  So you really don't even need the `String` class to hold on to a whole line.  BTW, `String` can make your program fail after random periods of time, but like I said, you don't even need it.

closed 2016-08-29 by SlashDevin