SlashDevin/NeoGPS #6: NeoGPS::Location_t::RAD_PER_DEG' cannot appear in a constant-expression

shrikantbarve, 2016-08-31:
Hello,

I am facing the issue while compiling on Arduino IDE. Here is exact error message >>

```
In file included from C:\Users\abc\Documents\Arduino\libraries\NeoGPS-master/GPSfix.h:34:0,
                 from C:\Users\abc\Documents\Arduino\libraries\NeoGPS-master/NMEAGPS.h:28,
                 from neogps_neoswserial.ino:1:
C:\Users\abc\Documents\Arduino\libraries\NeoGPS-master/Location.h:57:52: error: 'NeoGPS::Location_t::RAD_PER_DEG' cannot appear in a constant-expression
     CONST_CLASS_DATA float DEG_PER_RAD     = 1.0 / RAD_PER_DEG;
                                                    ^
Error compiling.
```

SlashDevin, 2016-08-31:
It's probably because you put NeoGPS in the Libraries/NeoGPS-master subdirectory.  They are supposed to go in your sketch directory.

shrikantbarve, 2016-09-02:
Hello,

Still it is not working, here is my directory structure >>

[image]

Once I click on Verify button, still it is giving me the same error:

[image]

shrikantbarve, 2016-09-02:
I followed Steps at

https://github.com/SlashDevin/NeoGPS/blob/master/doc/Installing.md

but still getting same error please find screenshots:

[image]

[image]

I am totally clueless :-(

shrikantbarve, 2016-09-02:
I was using Arduino 1.6.5 and after upgrading to 1.6.11 which is the current one as of now, it has successfully compiled.

SlashDevin, 2016-09-02:
I see.  It was a very obscure problem in that version of the IDE.  I have made a change to work around it in 1.6.5.

Thanks!

closed 2016-09-02 by SlashDevin