SlashDevin/NeoGPS #7: Cannot compile for Teensy 3.2

mjs513, 2016-09-06:
Hi,

First I just want to say great library and its been fun learning how to use it.  The forum helped a lot.  I am running windows 10 64bit, IDE 1.6.11 and teensyduino 1.30 beta 3.  Receiving following errors when I try using the Teensy3.2 using the NEMAcoherent sketch.  It does work fine for the Mega 2560.
> Arduino: 1.6.10 (Windows 10), TD: 1.30-beta3, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz optimize speed (overclock), US English"
> 
> Build options changed, rebuilding all
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseGSV(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:800:68: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::satellites[((int)((NMEAGPS*)this)->NMEAGPS::sat_count)].NMEAGPS::satellite_view_t::azimuth' to 'uint16_t& {aka short unsigned int&}'
> 
>                        parseInt( satellites[sat_count].azimuth, chr );
> 
>                                                                     ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseLat(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1269:9: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::location.NeoGPS::Location_t::_lat' to 'int32_t& {aka long int&}'
> 
>          );
> 
>          ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseLon(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1308:9: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::location.NeoGPS::Location_t::_lon' to 'int32_t& {aka long int&}'
> 
>          );
> 
>          ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseHDOP(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1417:40: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::hdop' to 'uint16_t& {aka short unsigned int&}'
> 
>      if (parseFloat( m_fix.hdop, chr, 3 ))
> 
>                                         ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parseVDOP(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1432:40: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::vdop' to 'uint16_t& {aka short unsigned int&}'
> 
>      if (parseFloat( m_fix.vdop, chr, 3 ))
> 
>                                         ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parsePDOP(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1447:40: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::pdop' to 'uint16_t& {aka short unsigned int&}'
> 
>      if (parseFloat( m_fix.pdop, chr, 3 ))
> 
>                                         ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parse_lat_err(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1462:46: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lat_err_cm' to 'uint16_t& {aka short unsigned int&}'
> 
>      if (parseFloat( m_fix.lat_err_cm, chr, 2 ))
> 
>                                               ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parse_lon_err(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1477:46: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::lon_err_cm' to 'uint16_t& {aka short unsigned int&}'
> 
>      if (parseFloat( m_fix.lon_err_cm, chr, 2 ))
> 
>                                               ^
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp: In member function 'bool NMEAGPS::parse_alt_err(char)':
> 
> C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS-master\NMEAGPS.cpp:1492:46: error: cannot bind packed field '((NMEAGPS*)this)->NMEAGPS::m_fix.gps_fix::alt_err_cm' to 'uint16_t& {aka short unsigned int&}'
> 
>      if (parseFloat( m_fix.alt_err_cm, chr, 2 ))

                                              ^

Error compiling for board Teensy 3.2 / 3.1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

SlashDevin, 2016-09-07:
Hmm, it looks like `NEOGPS_PACKED_DATA` is defined in NeoGPS_cfg.h.  Any chance you have an old version in your sketch directory?  Either get the latest version, or comment out that define.  It's also described [here](https://github.com/SlashDevin/NeoGPS/blob/master/doc/Configurations.md#platforms).

mjs513, 2016-09-07:
Just downloaded the lib a coiuple days go so Ithink I got the latest version. DIdnt read the platform.MD file until you just mentioned it.  That is probably the reason. Will give it a try first thing in the morning.  Thanks for getting back to so fast
Mike

closed 2016-09-07 by mjs513

SlashDevin, 2016-09-07:
Well, that's a little curious.  That define is guarded by
```
#ifdef __AVR__
  #define NEOGPS_PACKED_DATA
#endif
```
which should not be defined for the Teensy build (AVR is the 8-bit family of processors).  It might be a Teensy package error.  I'm checking over at the PJRC forum...

mjs513, 2016-09-07:
Just for sake of completeness since I was using the FULL config I copied those files over to the main lib per instructions.  I commented the  NEOGPS_PACKED_DATA  is defined in NeoGPS_cfg.h but I don't see the #ifdef guard you mentioned in your last post.  Also, from what I remember at some point the Teensy 3.x is identified as AVR during compile and if you want to put a guard in you have identify the boards themselves.  Don't hold me to this one though.

SlashDevin, 2016-09-08:
> since I was using the FULL config I copied those files
There we go!  Those files are out of date WRT the root.  For now, I would suggest taking the root files and modifying them yourself.  Thanks!

mjs513, 2016-09-09:
Just took care of that.  Thanks for update and warning.

Mike