SlashDevin/NeoGPS #8: ESP8266 Compile error

slave2anubis, 2016-09-11:
Hi, so i tried to compile one of the examples to work with a nodeMCU ESP8266 board, but it failed. Is there a way to make this lib compatibile with this board?

SlashDevin, 2016-09-11:
Yes, but I'll need to see the errors, and be sure to use the 3 config files (unmodified) that are in the root directory.

slave2anubis, 2016-09-11:
Well its somthing about avr/interupts and other files that are missing...

SlashDevin, 2016-09-11:
Ah, yes... I just checked in a few files that guard the AVR-specific includes.  The examples build when modified to use `Serial` for the `gps_port`:
```
HardwareSerial & gps_port = Serial;
#define GPS_PORT "Serial"
```
This assumes the GPS module TX pin is connected to the ESP8266 RX pin.  Also note that all the examples print information <b><i>to</i></b> `Serial`, so you should consider connecting the ESP8266 TX line to a serial-to-USB converter if you want to see those messages on your PC.  OTOH, you would not be able to send configuration messages to the GPS module.

Fair warning: I won't be able to test their execution.

slave2anubis, 2016-09-11:
Yes, but there is another error that i forgot about:
`osaCompat.h:4:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>`

SlashDevin, 2016-09-11:
Right, that was fixed.  That include isn't on line 4 any more.

slave2anubis, 2016-09-11:
OK, i got it to compile, but to use the softSerial you need to install a diferent lib from [here](https://github.com/plerup/espsoftwareserial). Any ideas how to test if i get any GPS info loss?

SlashDevin, 2016-09-11:
> Any ideas how to test if i get any GPS info loss?

Please try NMEAtest.ino.  It will verify that NMEA data will be parsed correctly, without requiring a GPS module.  It has some hard-coded NMEA data (in memory) that it tries to parse, and it only prints PASS/FAIL messages to Serial (you can change that to a different port).  You will have to uncomment several things in NMEAGPS_cfg.h and GPSfix_cfg.h before it will compile.  If it passes all the tests, then the library has been successfully ported to the ESP8266.  It would be safe to say that there would be no GPS info loss that was caused by parsing errors.

If you are asking about losing raw GPS characters coming in on a software or hardware serial port, then I would suggest trying each of the example sketches: NMEA, NMEAorder (check LAST_SENTENCE in NMEAGPS_cfg.h), NMEAloc, and NMEAcoherent.

I do not know your system connections, so you will have to choose the debug output port (typically `Serial`) and the gps input port (typically a NeoSWSerial on the defined RX_PIN and TX_PIN).  If you want to use the ESP8266 SoftwareSerial library, modify these lines at the top of each example program:

```
#else  
  // Only one serial port is available, uncomment one of the following:
  //#include <NeoICSerial.h>
  //#include <AltSoftSerial.h>
  //#include <NeoSWSerial.h> not on ESP8266!
  #include <SoftwareSerial.h>
#endif

// specify RX_PIN and TX_PIN here if the pins in GPSport.h (4 & 3) are not the ones you want:
#define RX_PIN 2
#define TX_PIN 3

#include "GPSport.h"
```
You can change the debug output port here:
```
#ifdef NeoHWSerial_h
  #define DEBUG_PORT NeoSerial
#else
  // Pick a different port here
  #define DEBUG_PORT Serial
#endif
```
If the example sketches show consistent data output (see Installation or Troubleshooting pages for example output), then they are not losing any GPS characters.  You should see one line of comma-separated values (CSV), once per second.  Consecutive lines should have nearly the same empty or populated values, depending on the signal quality.

I cannot over-emphasize that the example programs' loop structure is *crucial* to not losing GPS characters.  Spending too much time away from loop at the wrong time, in some other library, or by using `delay`, will most certainly cause character loss (i.e., input buffer overflow).  The Troubleshooting section describes several ways you can get into trouble.  :)

SlashDevin, 2016-09-12:
Oh, yes.  If you [enable NMEAGPS_STATS](https://github.com/SlashDevin/NeoGPS/blob/master/doc/Configurations.md#enabledisable-gathering-interface-statistics) in NMEAGPS_cfg.h, the example programs will display how many characters, sentences and CS errors have occurred.  As long as the characters and sentences are both increasing, and the CS errors are always zero, no GPS characters are being lost.

closed 2016-09-20 by SlashDevin