SlashDevin/NeoGPS #12: No option for UBX-NAV-DOP message in ubx.h/.cpp/ubxCFG.config

mjs513, 2016-09-24:
Think I figured out the issue based on your comments in issue #11.  Since I am running the ublox.ino example it uses the ubxGPS.cpp to parse the UBX messages (I can it commands previously) so I checked the ubxCFG file and ubxGPS.h/.cpp and there is no parsing option for the UBX-NAV-DOP message that contains those fields

mjs513, 2016-09-25:
Probably would add UBX-NAV-POSECEF as well.

closed 2016-09-25 by SlashDevin

SlashDevin, 2016-09-25:
I reconsidered this as a defect, not an enhancement, because there were `gps_fix` members that could have held the DOPs, if NAV-DOP were implemented.

The POSECEF request should be a new issue, if you'd like to request that.  It would require new members in `gps_fix`, so I would consider that an "enhancement".

mjs513, 2016-09-26:
Don't understand. If it is a defect why did you close the issue. Understand about about the other nav message.

Mike

SlashDevin, 2016-09-26:
> Don't understand. If it is a defect why did you close the issue. 

I tagged it as an enhancement after you created the issue, reasoning that it was "new functionality".  After thinking about it, I decided it was a bug (aka "defect") in the `ubloxGPS`, because *existing* fix members (DOPs) would never get a value.  This bug was fixed by code changes in the commit above, 6410139, and github automatically closes an issue if the commit comment says something like "Fix #12".

Closing an issue has nothing to do with the tags on the issue ("help wanted", "enhancement", "invalid", "question", etc.).  I could even tag an issue with "wontfix" and then close it.  The "open issues" are a kind of TO DO list, so most developers will close them to indicate that the issue is no longer receiving attention, regardless of its tags.

LOL, I just  noticed that there is a "bug" tag.  That would have been appropriate, and I'll be using that from now on.  So far, all my issues have been bugs, so I haven't been using that tag.

SlashDevin, 2016-10-15:
Sorry, it looks like I did not check in the updated ublox.ino that provided the DOPs...  Maybe that's what you were asking.  I just checked it in.

commit 6410139b (2016-09-25): Fix #12

* New NAV-DOP option in ubx_cfg.h
* Also default COMMA_NEEDED to enabled