SlashDevin/NeoGPS #14: Arduino Library Manager support

JohnWasser, 2016-09-28:
It would be nice if the library were available for installation through the Arduino Library Manager.  Here is the page about getting a library hosted on github into the list:

[https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ](url)

SlashDevin, 2016-09-28:
I have spent some time trying to get the IDE build process to properly resolve include file locations from inside library files.  For example, if this were in Libararies/mylib/mylib.h:
```
#include "sketch_directory_file.h"
#include <library_directory_file.h>
```
...the local "sketch_directory_file.h" is not copied to the build directory, and the compiler emits a "File not found" error.  :-(

In the NeoGPS context, there are several config files that should be in the sketch directory (i.e. a "user" or "local" include).  This allows different configurations for different sketches.  All the other NeoGPS files could be shared from the Libraries directory (via a <system> or <libraries> include), but there does not seem to be a way to do that...

The C standard actually says that [both include types are "implementation-defined"](http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1570.pdf#page=182) (i.e., I shouldn't expect anything).  The g++ documentation says that [it follows my expectation](https://gcc.gnu.org/onlinedocs/cpp/Include-Syntax.html).  Unfortunately, the IDE build process seems to break the g++ behavior.

I haven't tried to make NeoGPS a traditional Arduino IDE library since 1.0.5, so I'll see what I can get to work... maybe it already works.  :D   Or, if you know the magic incantation to make a library file include a sketch directory file, I'd be eternally grateful.

SlashDevin, 2016-10-22:
I have tried a few different things, and still cannot get a file in the Libraries subdirectory to include a file from the sketch directory.  Oh, well.  I've been considering moving it all to the Libraries subdirectory, but I'm reluctant to make it difficult for the occasional user who needs different configurations for different sketches.  Maybe I could organize it as a standard library and document how to have different configurations for the "expert".

SlashDevin, 2016-11-07:
Ok, ok, ok... :)  It's a standard library now, but I haven't pushed it into the Library Manager's list, yet.

SlashDevin, 2016-11-10:
NeoGPS is now available from the Arduino IDE Library Manager.

closed 2016-11-10 by SlashDevin