SlashDevin/NeoGPS #15: How to measure distance between two changing locations

dave-sausages, 2016-09-30:
Am I correct in thinking that the distance measuring function can only work to calculate the distance from the current to a location setup in the variable declared before void setup? I'd like to "save" a location at some point in my program, and then at a later stage measure the distance traveled from that point. This location will be changed by the user multiple times. And I'm sorry if this shouldn't be requested here, but I can't find a help or ask a question button anywhere.

SlashDevin, 2016-09-30:
Actually, the distance functions are members of the `Location_t` class (see Location.h), and they operate on instances of `Location_t`.  So you can calculate the distance between any two variables of the type `Location_t`.

The fixes returned from the NeoGPS have a `Location_t` inside them, so you can calculate the distance between two fixes' locations:
```
    dist = fix.location.DistanceKm( oldFix.location );
```
You would have had to save `oldFix` at some previous time.

Or, you can declare your own variables of type `Location_t`, and calculate the distances between them or any other locations (e.g., `fix.location`).  The example programs use one hard-coded location and the current `fix.location`, but you could also save the lat/lon at some time, then reload it into your own `Location_t` variable:

```
NeoGPS::Location_t base; // no value yet

void foo()
{
  int32_t prevLat = ???; // code to retrieve a long integer for the latitude
  int32_t prevLon = ???; // code to retrieve a long integer for the longitude

  base.lat( prevLat ); base.lon( prevLon ); // set the lat/lon members of the base Location

  float dist = base.DistanceKm( fix.location );
  //  you can also write
  //    dist = NeoGPS::Location_t::DistanceKm( base, fix.location );
}
```

ZlapiNunse, 2017-09-24:
I want, that when I turn on the device, the first thing which the device should do. It is identify the base location, and write this value the prevLat, prevLon (automaticaly) with the real localation in that moment.

I tried changing the following code of the last message (from SlashDevin):  
 int32_t prevLat = ???; // code to retrieve a long integer for the latitude
  int32_t prevLon = ???; // code to retrieve a long integer for the longitude

for this, I wrote:

      int32_t prevLat = fix.latitude(); // code to retrieve a long integer for the latitude
      int32_t prevLon = fix.longitude(); // code to retrieve a long integer for the longitude
It compiling well, but when I watch the Serial Monitor, appear a none logic result. The result in my localation is 4683.63. Which it don't make sense, because If I turn on the device, and I remain quit ( I don't move) the result I hope is 0. It should increase, as far as I am far, from my base. 

Can somebody help me? Please, I am desperate.

I HOPE YOUR ANSWER.
Thank you in advance. 

I wrote my whole code here:

#include <NMEAGPS.h>

//======================================================================
//  Program: NMEAdistance.ino
//
//  Description:  Display distance from a base location.
//
//  Prerequisites:
//     1) NMEA.ino works with your device (correct TX/RX pins and baud rate)
//     2) GPS_FIX_LOCATION has been enabled.
//     3) A sentence that contains lat/long has been enabled (GGA, GLL or RMC).
//     4) Your device sends at least one of those sentences.
//
//  'Serial' is for debug output to the Serial Monitor window.
//
//  License:
//    Copyright (C) 2014-2017, SlashDevin
//
//    This file is part of NeoGPS
//
//    NeoGPS is free software: you can redistribute it and/or modify
//    it under the terms of the GNU General Public License as published by
//    the Free Software Foundation, either version 3 of the License, or
//    (at your option) any later version.
//
//    NeoGPS is distributed in the hope that it will be useful,
//    but WITHOUT ANY WARRANTY; without even the implied warranty of
//    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
//    GNU General Public License for more details.
//
//    You should have received a copy of the GNU General Public License
//    along with NeoGPS.  If not, see <http://www.gnu.org/licenses/>.
//
//======================================================================

#include <GPSport.h>

#include <Streamers.h>

//------------------------------------------------------------
// Check that the config files are set up properly

#if !defined( NMEAGPS_PARSE_RMC ) &  \
    !defined( NMEAGPS_PARSE_GGA ) &  \
    !defined( NMEAGPS_PARSE_GLL )
#error You must uncomment at least one of NMEAGPS_PARSE_RMC, NMEAGPS_PARSE_GGA or NMEAGPS_PARSE_GLL in NMEAGPS_cfg.h!
#endif

#if !defined( GPS_FIX_LOCATION )
#error You must uncomment GPS_FIX_LOCATION in GPSfix_cfg.h!
#endif

//------------------------------------------------------------
//  Define a set of GPS fix information.  It will
//  hold on to the various pieces as they are received from
//  an RMC sentence.  It can be used anywhere in your sketch.
static gps_fix  fix;

NMEAGPS gps;

// The base location, in degrees * 10,000,000NeoGPS::Location_t base; // no value yet
NeoGPS::Location_t base( fix.latitude(), fix.longitude() );
bool stScan = true;
void setup()
{
  DEBUG_PORT.begin(9600);
  DEBUG_PORT.println( F("NMEAdistance.ino started.") );
  DEBUG_PORT.println( F("Looking for GPS device on " GPS_PORT_NAME) );
  gpsPort.begin(9600);
}
void loop()
{
  while (gps.available( gpsPort )) {
    gps_fix fix = gps.read(); // save the latest
    if (fix.valid.location) {
      if (stScan==true){
      int32_t prevLat = fix.latitude(); // code to retrieve a long integer for the latitude
      int32_t prevLon = fix.longitude(); // code to retrieve a long integer for the longitude
      base.lat( prevLat ); base.lon( prevLon ); // set the lat/lon members of the base Location
      stScan=false;
      }
      float dist = base.DistanceKm( fix.location );
      DEBUG_PORT.print( F("DisntanceKm: ") );
      DEBUG_PORT.println( dist );
      trace_all( DEBUG_PORT, gps, fix );
    } else
      trace_all( DEBUG_PORT, gps, fix );
  }
} // loop

SlashDevin, 2017-09-25:
Just use the `stScan` flag to set the `base` value the first time:

```
NeoGPS::Location_t base;
bool stScan = true;

void setup()
{
  DEBUG_PORT.begin(9600);
  DEBUG_PORT.println( F("NMEAdistance.ino started.") );
  DEBUG_PORT.println( F("Looking for GPS device on " GPS_PORT_NAME) );
  gpsPort.begin(9600);
}

void loop()
{
  while (gps.available( gpsPort )) {
    gps_fix fix = gps.read(); // save the latest

    if (fix.valid.location) {
      if (stScan){
        stScan = false;
        base   = fix.location;
      } else {
        float dist = base.DistanceKm( fix.location );
        DEBUG_PORT.print( F("Distance Km: ") );
        DEBUG_PORT.println( dist );
      }
    }

    trace_all( DEBUG_PORT, gps, fix );
  }
} // loop
```
You can also use the flag to know if you can calculate (and print) the distance from the `base`.