SlashDevin/NeoGPS #16: Traveled Distance?

pcace, 2016-10-11:
Hey, thank you so much for all the work! Great library (even kind of usable for me as a total beginner;)!
But i have to questions: 
1. i cannot find a simple way to get the traveled distance. Is there anything already implemented in the library, what i could use to realize some kind of odometer?

2. I am working right now with the example file NMEASDLog.ino. It workes wonderfully, i changed a little bit, that i directly outputs gpx files, so nothing special. But here is the question: if i not only want to write down anything to SD card, but also display it on an lcd, where would i put the additional code?

Thanks a lot!,

Pcace

SlashDevin, 2016-10-12:
> **1)**  i cannot find a simple way to get the traveled distance. Is there anything already implemented in the library, what i could use to realize some kind of odometer?

No, there is no odometer "function".  However, there is a distance function, so you could calculate a distance from a base to the current location.  That would just be the "air" distance, and would not act like an odometer.

Or you could accumulate distances into a total distance traveled, which would act like an odometer.  Simply calculate the distance between a previous point and the current point, and add it to the total.  Then set the previous point to the current point.

You *could* use the previous fix and the current fix, but that will accumulate distance even if you're not moving (GPS jitter).  Because there are many different ways to pick locations for accumulating distance, an odometer function is not implemented in NeoGPS.

You should probably pick something that matches the behavior of your platform.  Are you walking?  Then you should probably average points for 15 seconds and calculate the distance from the previous 15-second interval.  If it's in a car, try a 5 second window and don't accumulate distance if the speed is less than 10kph.

> **2)** I am working right now with the example file NMEASDLog.ino... if i want to display it on an lcd... the very last [part] is:
```
  myGLCD.print("TEST", 0, 0);
}
```
But it is not working at all. (it simply does not show anything). also when adding a: DEBUG_PORT.println( F("DEBUGTEST") ); it does not show up.
Is there any reason, that commands in loop() are not dealt with?

Yes.  Please read the [Troubleshooting](https://github.com/SlashDevin/NeoGPS/blob/master/doc/Troubleshooting.md) section, especially part 5, "Trying to do too many things".  You need to realize that one character at 9600 takes 1ms to be received (from the GPS) or transmitted (to the Serial Monitor or the TFT).  Although that's very fast compared to the human timescale, the `loop` function will get called tens of thousands of times.

This means that you are trying to print "TEST" as fast as the Arduino can do it (about 250 times per second).  While the Arduino is trying to print, the GPS is still sending.  Eventually, the GPS will overflow the input buffer or, in the case of the NMEASDlog sketch, the fix buffer will fill (aka data overrun).  New GPS data (or fixes) will be discarded.  So... don't do that.  You can't update the TFT 250 times per second, either.

The only safe time to do something expensive (like update the TFT) is immediately after a fix is received.  In most of the example programs, this would be in the `doSomeWork` function.  In the NMEASDlog sketch, it would be in the `while` loop, sometime after the `gps.read()` on [line 210](https://github.com/SlashDevin/NeoGPS/blob/master/examples/NMEASDlog/NMEASDlog.ino#L210).  You can do it before or after the SD file is written.

You still have to be careful about how much you do.  If the TFT update takes 2 seconds (I doubt that), then you can't update the TFT on every update.  Do it every 3rd fix instead.  Otherwise, the fix buffer will fill up before you can get to the fixes.

You may need to measure the TFT update time.  Save the `millis()` value before you start, then subtract it from `millis()` value when the TFT is fully updated and print it out.  (This is the same technique used to determine the SD logging time, [here](https://github.com/SlashDevin/NeoGPS/blob/master/examples/NMEASDlog/NMEASDlog.ino#L216).) Some SD writes can take 100ms, so you have to account for that time, too.

You can increase the FIX_MAX to accommodate a long TFT update time, but you cannot spend more time than an *average* of 1s for TFT + SD log times.  (This assumes a 1Hz update rate from your GPS.)

pcace, 2016-10-14:
Hey, thank you so much for your very detailed answer!!!
I did it like this for now: 

>   if (gps.available()) {
>     gps_fix fix = gps.read();
>
>     //GPS DISTANCE
>     if (fix.valid.location) {
> 
>       if (flt == 0) {
>         flt = 1;
>         pos_alt = fix.location;
>       }
>       dist_alt = dist;
>       dist = fix.location.DistanceKm( fix.location, pos_alt );
>       pos_alt = fix.location;
>       dist = dist + dist_alt;
>     }
> 
>     myGLCD.print("DST: " + String(dist) + " KM", 0, 0);
>     myGLCD.print("LAT: " + String(fix.latitudeL()), 0, 30);
>     myGLCD.print("LON: " + String(fix.longitudeL()), 0, 60);
>     myGLCD.print("ALT: " + String(fix.altitude()) + " M", 0, 90);
>     myGLCD.print("SPD: " + String(fix.speed_kph()) + " KM/H", 0, 120);
>     myGLCD.print("SAT: " + String(fix.satellites), 0, 150);
>     myGLCD.print("DOP: " + String(fix.hdop), 0, 180);
>   }

which kind of works ;) - but it only shows anything on the display if i uncomment waitForFix(); no matter if i have fix or not. i really tried to understand why this happens, but i simply do not get it. 
It shows and saves now pretty good datasets. 

I hope i understand your answer correctly, but just to be sure, i should put the whole (section above) into a millis loop, to just fire it up once a second right? - to make sure i do not get into the SD save and GPS read timing?

Thank you so much (again!) for the great library, and sorry for these noob-questions ;)

Pcace

SlashDevin, 2016-10-14:
Ok, first... please don't use the `String` class.  There are so many random problems it can cause, and I'd rather not worry about whether `String` is causing a problem now.  :P  You can just do this: 
```
     //PRINT TO LCD
     myGLCD.print( F("DST: "), 0, 0 );
     myGLCD.print( dist );
     myGLCD.print( F(" KM") );

     myGLCD.print( F("LAT: "), 0, 30);
     myGLCD.print( fix.latitudeL() );

     myGLCD.print( F("LON: "), 0, 60 );
     myGLCD.print( fix.longitudeL() );

       etc.
```
**BONUS:** Your program will use 1600 fewer bytes!  If your LCD library doesn't like the F macro around the "labels", just leave it out.

> it only shows anything on the display if i uncomment waitForFix(); no matter if i have fix or not.

Do you mean commenting out the `waitForFix()` at the end of `setup()`?  I don't see anything in `waitForFix()` that is required...  Did you modify `waitForFix()`?  Try the above changes, then just comment and uncomment that call in `setup()` and tell me what happens.  That will eliminate a possible `String` problem.

> i should put the whole (section above) into a millis loop, to just fire it up once a second right? - to make sure i do not get into the SD save and GPS read timing?

Well, the `gps.available()` will only occur once per second.  That's a *very* accurate second, much more accurate than `millis()`.  So the nested GLCD prints are already happening at that rate.

The real question is, "Are you are getting DATA_OVERRUN messages?"  If you are, then the SD write time plus the GLCD prints is taking too long, and you're losing some fixes.  If you aren't getting those messages, then there isn't a problem.

pcace, 2016-10-15:
Hey, Thank you again for your help!!! It seems, that i have to use the String() thing in order to get the Display to show something, if i remove it, or simply try 
> myGLCD.print( dist , 0,0 );
the error is: 
>    void print(String st, int x, int y, int deg=0);

2. I did not changed anything in the waitForFix(); funtion, but still, if i simply uncomment it in the Setup(); it works. If i dont, the display doesnt show anything... strange but hey ;)

3. i do get 
> DATA OVERRUN: fix data lost!
not often, but from time to time (2 times a minute maybe. I did not yet measured the time it needs to write to the SD card, but i assume itl take much longer than expected ;) The routine looks like this: 

>    if (fix.altitude() != 0 && fix.satellites > 3) {
> 
>         logfile.println("<trkseg>");
>         logfile.print("<trkpt lat=\"");
>         printL( logfile, fix.latitudeL() );
>         logfile.print( "\" lon=\"" );
>         printL( logfile, fix.longitudeL() );
>         logfile.print("\">");
> 
>         logfile.println("");
> 
>         logfile.print("<ele>");
>         logfile.print(fix.altitude());
>         logfile.print("</ele>");
> 
>         logfile.println("");
> 
>         logfile.print("<speed>");
>         logfile.print(fix.speed_kph());
>         logfile.print("</speed>");
> 
>         logfile.println("");
> 
>         logfile.print("<course>");
>         logfile.print(fix.heading_cd());
>         logfile.print("</course>");
> 
>         logfile.println("");
> 
>         logfile.print("<time>20");
>         logfile.print(fix.dateTime.year);
>         logfile.print("-");
>         if (fix.dateTime.month < 10)
>           logfile.print( '0' );
>         logfile.print(fix.dateTime.month);
>         logfile.print("-");
>         if (fix.dateTime.day < 10)
>           logfile.print( '0' );
>         logfile.print(fix.dateTime.day);
>         logfile.print("T");
>         if (fix.dateTime.hours < 10)
>           logfile.print( '0' );
>         logfile.print(fix.dateTime.hours);
>         logfile.print( ':' );
>         if (fix.dateTime.minutes < 10)
>           logfile.print( '0' );
>         logfile.print(fix.dateTime.minutes);
>         logfile.print( ':' );
>         if (fix.dateTime.seconds < 10)
>           logfile.print( '0' );
>         logfile.print(fix.dateTime.seconds);
>         logfile.print( '.' );
>         if (fix.dateTime_cs < 10)
>           logfile.print( '0' ); // leading zero for .05, for example
>         logfile.print(fix.dateTime_cs);
>         logfile.print("Z</time>");
> 
>         logfile.println("");
> 
>         logfile.print("<sat>");
>         logfile.print(fix.satellites);
>         logfile.print("</sat>");
> 
>         logfile.println("");
> ....

+ a couple of static lines in Setup(); (setting the GPX header in the file

So i guess for every println(); it is doing a seperate write to the SD card? that is probably totally killing the timing... right? As an idea: creating a function wich collects all the data and then writes everything in one go?

Thank you again for your help!!

pcace

SlashDevin, 2016-10-17:
> I have to use the String() thing in order to get the Display to show something, if i remove it, or simply try 
> &nbsp;&nbsp;&nbsp;&nbsp;`myGLCD.print( dist , 0,0 );`
> the error is: 
> >  void print(String st, int x, int y, int deg=0);

That's not the error, but I suspect it says something like "no matching function".  And that's not what I suggested above.  This should work:
```
     //PRINT TO LCD
     myGLCD.setcursor( 0, 0 );
     myGLCD.print( F("DST: ") );
     myGLCD.print( dist );
     myGLCD.print( F(" KM") );

     myGLCD.setcursor( 0, 30 );
     myGLCD.print( F("LAT: ") );
     myGLCD.print( fix.latitudeL() );

     myGLCD.setcursor( 0, 60 );
     myGLCD.print( F("LON: ") );
     myGLCD.print( fix.longitudeL() );
```
Using the F macro will save a bunch of RAM.  You should do it for all those `logfile` prints:
```
logfile.print( F("<trkpt lat=\"") );
```
All the prints with a "double quoted" string should use the F macro.  Lower-case "string" is **not** the `String` class.  These are `char` arrays, also called "C strings".

BTW, which LCD library are you using?  There are several different ones, and they have slightly different features.  I really can't see your Arduino directory...  :)

> I do get DATA OVERRUN, not often, but 2 times a minute maybe. I did not yet measure the time it needs to write to the SD card, but I assume it take much longer than expected.

The original NMEASDlog.ino has a variable [here](https://github.com/SlashDevin/NeoGPS/blob/master/examples/NMEASDlog/NMEASDlog.ino#L216) that is used to log the time [here](https://github.com/SlashDevin/NeoGPS/blob/master/examples/NMEASDlog/NMEASDlog.ino#L245).  If you grab the time at the top:
```
    if (fix.altitude() != 0 && fix.satellites > 3) {
      static uint16_t lastLoggingTime  = 0;
             uint16_t startLoggingTime = millis();
```
...  then update the LCD and write to the logfile, then write the loggingTime:
```
      logfile.print( lastLoggingTime ); // write how long the previous logging took
      logfile.println();
```
...  you'll know how long *both* things take.  Run the sketch long enough to get a few overruns and look at the logging time in the SD file.  I suspect you will need to increase FIX_MAX by one; instructions for using the logging time to determine your fix max are [here](https://github.com/SlashDevin/NeoGPS/blob/master/examples/NMEASDlog/NMEASDlog.ino#L159).

You can also increase the flush time, depending on how much data you would be willing to lose.  Read comments [here](https://github.com/SlashDevin/NeoGPS/blob/master/examples/NMEASDlog/NMEASDlog.ino#L248).  Some people use a button press or  a command from the Serial Monitor window (if you're attached to a PC) to end the sketch with:
```
    logfile.close()
    for (;;);  // hang!
```
This guarantees you don't lose the last set of data.

> for every println(); it is doing a separate write to the SD card? that is probably totally killing the timing... right? As an idea: creating a function which collects all the data and then writes everything in one go?

Nope, the SD card is already buffering everything for you.  Most prints will just put a few characters in the internal SD buffer.  Once or twice per fix, the 512-byte buffer will get written to the SD card (i.e., "flushed").

It is important for you to get rid of the `String` class and use the F macro.  Until then, we don't know if you're actually fighting a memory corruption problem or low memory problem.  :(

closed 2016-10-22 by SlashDevin