SlashDevin/NeoGPS #18: Compile error on Arduino Mega

riccardopersello, 2016-11-27:
Hello,
I've tried to compile NMEA.ino, NMEAdiagnostics.ino, NMEAtest.ino and ublox.ino but compilation fails with these errors:  
`
In file included from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:22:0from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:29,
                 from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\examples\NMEA\NMEA.ino:2:
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: error: 'constexpr' does not name a type
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:137:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA float KM_PER_NMI = 1.852;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:137:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA float KM_PER_NMI = 1.852;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: error: 'constexpr' does not name a type
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:140:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA uint16_t M_PER_NMI = 1852;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:140:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA uint16_t M_PER_NMI = 1852;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: error: 'constexpr' does not name a type
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:143:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA float MI_PER_NMI = 1.150779;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:143:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA float MI_PER_NMI = 1.150779;
     ^
In file included from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:29:0,
                 from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\examples\NMEA\NMEA.ino:2:
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h: In member function 'float gps_fix::speed_kph() const':
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:138:52: error: 'KM_PER_NMI' was not declared in this scope
     float    speed_kph () const { return speed() * KM_PER_NMI; };
                                                    ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h: In member function 'uint32_t gps_fix::speed_metersph() const':
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:141:59: error: 'M_PER_NMI' was not declared in this scope
     uint32_t speed_metersph() const { return (spd.whole * M_PER_NMI) + (spd.frac * M_PER_NMI)/1000; };
                                                           ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h: In member function 'float gps_fix::speed_mph() const':
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:144:49: error: 'MI_PER_NMI' was not declared in this scope
     float  speed_mph() const { return speed() * MI_PER_NMI; };
                                                 ^
In file included from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:22:0,
                 from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:29,
                 from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\examples\NMEA\NMEA.ino:2:
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h: At global scope:
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: error: 'constexpr' does not name a type
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:96:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA nmea_msg_t NMEA_FIRST_MSG = (nmea_msg_t) (NMEA_UNKNOWN+1);
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:96:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA nmea_msg_t NMEA_FIRST_MSG = (nmea_msg_t) (NMEA_UNKNOWN+1);
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: error: 'constexpr' does not name a type
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:97:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA nmea_msg_t NMEA_LAST_MSG  = (nmea_msg_t) (NMEAMSG_END-1);
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:97:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA nmea_msg_t NMEA_LAST_MSG  = (nmea_msg_t) (NMEAMSG_END-1);
     ^
In file included from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\examples\NMEA\NMEA.ino:2:0:
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:290:41: error: 'NMEAGPS_KEEP_NEWEST_FIXES' was not declared in this scope
     static const bool keepNewestFixes = NMEAGPS_KEEP_NEWEST_FIXES;
                                         ^
In file included from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/GPSfix.h:22:0,
                 from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:29,
                 from C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\examples\NMEA\NMEA.ino:2:
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: error: 'constexpr' does not name a type
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:362:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA uint8_t NMEA_FIRST_STATE = NMEA_IDLE;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:362:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA uint8_t NMEA_FIRST_STATE = NMEA_IDLE;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: error: 'constexpr' does not name a type
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:363:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA uint8_t NMEA_LAST_STATE  = NMEA_RECEIVING_CRC;
     ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:81:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
   #define CONST_CLASS_DATA static constexpr
                                   ^
C:\Users\****USER****\Documents\Arduino\libraries\NeoGPS\src/NMEAGPS.h:363:5: note: in expansion of macro 'CONST_CLASS_DATA'
     CONST_CLASS_DATA uint8_t NMEA_LAST_STATE  = NMEA_RECEIVING_CRC;
     ^
exit status 1
Errore durante la compilazione per la scheda Arduino/Genuino Mega or Mega 2560.
`

But with Arduino Uno works fine...
Sorry for my bad English.

SlashDevin, 2016-11-29:
Which IDE version are you using?  If it is 1.0.6 or 1.0.5, a new version of the IDE will fix this.  Can you upgrade?

riccardopersello, 2016-11-29:
I use Arduino (.org) 1.7.11 on Mac, but I tried also arduino.cc 1.0.5 on Mac and 1.7.13 on Windows (also the Windows Store version) but it does not work.

riccardopersello, 2016-11-29:
Solved using the latest Arduino cc on Mac, I don’t recommend to compile it with the arduino.org version.

closed 2016-11-29 by riccardopersello

SlashDevin, 2016-11-29:
I just tested Arduino.cc 1.6.8 and 1.6.13, and they seem to work.  I have another report for Arduino.org 1.7.11 that it is not working.  I'll see if I can accommodate them both.  >:-|

SlashDevin, 2016-11-29:
Yes, the two branches have diverged.  I just checked in a NeoGPS_cfg.h that works under 1.7.x

FarazArdu, 2018-10-13:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
In file included from C:\Program Files (x86)\Arduino\libraries\NeoGPS\src/GPSfix.h:21:0,

                 from C:\Program Files (x86)\Arduino\libraries\NeoGPS\src/NMEAGPS.h:28,

                 from C:\Users\HP 15\Documents\sketch_oct13a\sketch_oct13a.ino:1:

C:\Program Files (x86)\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:106:35: error: 'constexpr' does not name a type

   #define CONST_CLASS_DATA static constexpr

                                   ^

C:\Program Files (x86)\Arduino\libraries\NeoGPS\src/Location.h:30:5: note: in expansion of macro 'CONST_CLASS_DATA'

     CONST_CLASS_DATA float LOC_SCALE = 1.0e-7;

     ^

C:\Program Files (x86)\Arduino\libraries\NeoGPS\src/NeoGPS_cfg.h:106:35: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11

   #define CONST_CLASS_DATA static constexpr