SlashDevin/NeoGPS #19: Problems using NMEAtimezone.ino example

TechStuffGuy, 2017-01-20:
Hi!
I use Arduino UNO and Neo m8n clone and like to control my scrolling clock with NMEAtimezone code. But when I want to see Daylight Savings Time added in real time I can't see it. For this I open **NMEAtimezone.ino** and uncomment **<AltSoftSerial.h>** and change my **zone_hours** and **changeover time**. First I noticed I must set changeover time in UTC and not by my local clock. But when I watch Serial Monitor and time passes my changeover time - nothing happening, time just ticking forward. Change in hours occur after Arduino reset. Somehow in my case **zone_hours++** not processed without reloading whole code. What I must change here?

SlashDevin, 2017-01-20:
> I must set changeover time in UTC and not by my local clock

Yes, you're right!  That is a bug.  The changeover time should be compared with the local time, not the UTC time.  I'll take a look.

closed 2017-01-20 by SlashDevin

TechStuffGuy, 2017-01-20:
Thanks, now all is working fine!

commit b6793566 (2017-01-20): Fix #19, timezone & DST offsets

* DST compare with local TZ, not UTC
* Fall DST changeover includes +1 for "current" DST offset