SlashDevin/NeoGPS #21: Satellite information

SebastianZug, 2017-02-11:
Hi SlashDevin,
I am trying to map the current satellite configuration to a reached precision level. Hence, I used the "full" configuration of NeoGPS to generate the required data based on a SainSmart Ublox NEO-6M module. But I realized that the "Satellite Array" included up to 16 satellites.  I believe to know, that more than 12 satellites should never be visible from a specific point. 

The documentation of gps.satellies[] mentions that : "There is additional information that is not related to a fix." Does the satellite array collects and stores the satellite parameters in a specific way? Are these information suitable for my project?

Best wishes 

Sebastian

SlashDevin, 2017-02-12:
> But I realized that the "Satellite Array" included up to 16 satellites. I believe to know, that more than 12 satellites should never be visible from a specific point.

I'm not sure what you are asking.  Many GPS devices claim to track up to 16 satellites at a time, but I've never seen more than 10.  In the default configuration, the `gps.satellites` array can hold information for 20 satellites.  You are free to modify the configuration item `NMEAGPS_MAX_SATELLITES` to a smaller number (see NMEAGPS_cfg.h).

> Does the satellite array collects and stores the satellite parameters in a specific way?

According to the [Choosing Your Configuration](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Choosing.md) page, the satellite array will be populated by GSA (satellite id only) and GSV (id, azimuth, elevation, signal strength and a "tracked" flag that indicates the GPS device is using it for the navigation solution).  Those are the only two sentences that will fill out the array.  You must enable `NMEAGPS_PARSE_GSA` and/or `NMEAGPS_PARSE_GSV` **and** your device must send them.  If it does not send them by default, you must ask for them or send a configuration command (the NEO-6M can do either).

> Are these information suitable for my project?

I'm not sure.  If by "trying to map the current satellite configuration", you mean that you are trying to draw a picture of the satellites in the sky, you must use the GSV sentence to get azimuth and elevation of each tracked satellite.  

If by "a reached precision level", you mean the precision *reported* by the GPS device, you would need the Degrees Of Precision (DOPs) and/or Error Estimates (latitude, longitude and altitude).  They can be obtained by enabling the GSA and/or GST sentences, respectively.

If you have a stationary platform and you are comparing the reported position to the actual position, you would not need the DOPs nor Error Estimates.

SebastianZug, 2017-02-12:
Hi SlashDevin,

thanks for the fast response and the detailed explanation! I run a measurement for approximatly 12 hours and received up to 16! available sensors and a permanently alternating number of satellites for different phases. According to 

[image]

I got 

```
"2017-02-08 21:32:32" 
'23 7/324@27,25 45/125@49,26 40/297@48,29 76/59@46,31 50/241@52,33 25/215@-,39 30/165@34,'

2017-02-08 21:32:33
'2 13/43@28,4 53/297@45,5 20/72@48,9 1/359@46,12 6/120@15,16 12/299@36,20 15/123@31,21 35/188@34,23 7/324@27,25 45/125@49,26 40/297@48,29 76/60@46,31 50/241@52,33 25/215@-,39 30/165@33,'

2017-02-08 21:32:34
'23 7/324@28,25 45/125@49,26 40/297@48,29 76/60@46,31 50/241@52,33 25/215@-,39 30/165@34,'

... 
```
the ossiziation between 7 and 15 points is visible for 30 minutes and I have no idea whats going wrong. I used the 'full' configuration and the 'traceall' function.

Looking forward to your remarks

Sebastian

SlashDevin, 2017-02-13:
There can be subtle interactions between the GSA and GSV, but I suspect that you are trying to [print too much](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Troubleshooting.md#printing-too-much).

The full configuration plus all the satellites' info is more than 200 characters per second, so the Arduino is "blocked" while the 1st to 136th characters are going out (the output buffer will hold the last 64 characters).    While it's blocked, GPS data may continue to arrive.  After the 64th character, the input buffer is full and no more characters will be saved for processing.  You may be losing the last two GSV sentences while printing the previous fix and satellites array.

A quick test would be to set your `Serial` baud rate to 115200.  That's just a guess, because I can't see your sketch.  If you embed it in your next comment, I could review it.

This also depends on the GPS baud rate and how many sentences it is sending.  If it sending "a lot" of sentences at 9600, the [quiet time](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Troubleshooting.md#quiet-time-interval) may be too short to print everything.  Having the wrong LAST_SENTENCE could also cause trouble (use NMEAorder.ino to confirm your choice for LAST_SENTENCE).

closed 2017-02-25 by SlashDevin