SlashDevin/NeoGPS #24: SBAS for better accuracy

tauco, 2017-03-01:
Hi SlashDevin!
I would like to know, how to get the SBAS(WASS, EGNO...) information. I haven't being able to find a way to do it in your libraries. 
Thanks in advance

SlashDevin, 2017-03-01:
If I understand your question, this is actually a function of the GPS device.  The SBAS information is used to improve the GPS information.  Several GPS devices allow you to enable or disable the use of SBAS, so you'll have to find the specific command for that.

If you are asking about retrieving the SBAS information being used by the GPS device, you will have to look at the specification for your device.  For example, you can query the ublox NEO-6M device for the pseudo range and ionosphere corrections with a binary message.  This requires using the binary UBX protocol classes of NeoGPS.

A different device would have a different query and response.

SlashDevin, 2017-03-01:
I'm still uncertain.  Do you 

1)  Want to enable or disable SBAS augmentation on the NEO-6M? You may have to send a binary command, but you could still receive NMEA fixes.

  or

2)  Do you want to query the SBAS corrections from the NEO-6M?  You would have to send a binary command, receive a binary response, but you could still receive NMEA fixes.  NeoGPS can parse both NMEA (text) and UBX (binary) messages at the same time.

tauco, 2017-03-01:
I want to do the second one, sorry for not explaining well. I thought you couldn't parse both at the same time. Now that you explained me and read your documentation more carefully I realize how to proceed   
Thank you very much. It is cristal clear now.

closed 2017-03-02 by SlashDevin