SlashDevin/NeoGPS #25: library not recognized in visual studio community edition 2017

xenek, 2017-04-29:
Hi,

Using Visual Micro Arduino IDE for visual studio community edition 2017,

Your library isn't recognized.

When installed from zip or from the library manager.

#include <NMEAGPS.h>
results in 'cannot open source file'
even though the library manager shows NeoGPS is installed.
I can't select it in the list of installed libraries, it doesn't show.

It is recognized in the Arduino compiler 1.8.2

Let me know if I need to push this to Visual Micro.
Cheers,
X

SlashDevin, 2017-04-29:
Hmm...  Are you sure you have the required permissions to add a library to Visual Micro?

Can other libraries be included without error?

Can other NeoGPS files be included?  Try `#include <NeoGPS_cfg.h>`.

NeoGPS is a little different, because there is no NeoGPS.h.  Try renaming the Libraries\NeoGPS subdirectory to Libraries\NMEAGPS.  That would make the NeoGPS library include filename match the subdirectory name.  I'm skeptical, because you say it is not in the "list of installed libraries".

The `library.properties` file identifies the includes to be added when you select `Add Library` in the Arduino IDE, but I do not know if there is some other interaction when using the Visual Studio IDE.  The latest versions of the Arduino IDE require all source code to be in a `src` subdirectory.

Have you tried any other "new" libraries that have source in the `src` subdirectory?

I would guess that this is almost certainly an installation or Visual Micro problem.  Please let me know if there's something I should do to make it compatible.

xenek, 2017-04-30:
I should have the required permissions - Visual Studio is set to run as Administrator.  Other libraries install correctly. As a test I clicked on the Adafruit GPS library. It installed correctly the first time, without error. A simple click on 'rescan' was all that was required for the Adafruit library to appear in the list of installed libraries. I can with a click remove it, and reinstall it with no error.

NeoGPS however doesn't show in the list of installed libraries, which is the default list that loads when running the Visual Micro board and library manager.

[image]

It does however show in the adjacent list as installed under 'manage libraries'.

[image]

And can be installed or removed to that list, on that tab without error. This toggles it between appearing black or shaded.

When installing it shows the version, and upon clicking, no errors are shown. However even after a rescan it doesn't appear as installed - the count remains at 39 user libraries rather than incrementing to 40.

[image]

The type of IDE is set to default, and the Visual Micro settings aren't customized. 

[image]

The files all look like they are there, even as Visual Studio reports that it can't open them.

[image]

The file in question is not corrupted and can be opened by other apps (and can be written to)

[image]

And the file can be manually opened in Visual Studio, showing and allowing further modification.

[image]

The things network library is a bit new, and has an SRC folder containing it's files. It seems OK and compares with yours if I check the library.properties file. Yet it shows and yours doesn't.

name=TheThingsNetwork
version=2.4.0
author=The Things Network
maintainer=Johan Stokking <johan@thethingsnetwork.org>
sentence=The Things Network Arduino Library.
paragraph=Compatible with any Microchip RN2483 and RN2903 device.
category=Communication
url=https://github.com/TheThingsNetwork/arduino-device-lib
architectures=*
includes=TheThingsNetwork.h

name=NeoGPS
version=4.1.3
author=SlashDevin
maintainer=SlashDevin
sentence=NMEA and ublox GPS parser, configurable to use as few as 10 bytes of RAM
paragraph=Faster and smaller than all other GPS parsers
category=Communication
url=https://github.com/SlashDevin/NeoGPS
architectures=avr,samd,sam
includes=NMEAGPS.h

So I am a bit stumped. I guess I'll repost the issue / check the Visual Micro forums further. Thanks for asking your questions.

UPDATE: I fixed it. Looking at the properties above and comparing I realized that architectures is a * on the things network library. I changed your architectures line to match TheThingsNetwork (Just a * rather than avr,samd,sam) Did a rescan, and now it shows.  Hopefully this helps someone :)

Cheers,
X

xenek, 2017-04-30:
Probably a Visual Micro Library Manager bug. Have listed it here. Please leave this bug report on GitHub up for at least a few days or so to give the admins at Visual Micro a chance to review my notes. Thanks!

http://www.visualmicro.com/forums/YaBB.pl?num=1493515806/0#0

xenek, 2017-04-30:
Ok, so Tim at Visual Micro has confirmed this:-

Thanks for the message.

You should let the library author know to add esp8266 to the list or change the architectures to * as you did. It's right that the library is ignored by visual micro and arduino ide for esp8266.

avr,samd,sam,esp8266 

btw: Not sure if you hit this but visual micro appeared to be giving a useless error when the architecture could not be found. It should be improved in the next release due shortly.

SlashDevin, 2017-04-30:
LOL, after realizing you were using ESP8266, I had to wonder about the `architectures` property.  It is not documented very well, IMHO.  I'll add esp8266.

closed 2017-04-30 by SlashDevin

SlashDevin, 2017-05-13:
Sorry, I just noticed the the library.properties file was not checked in.  Release v4.1.7 has the esp8266 architecture.  `arc32` was also added for the Arduino 101.

commit 781744ca (2017-04-30): Fix #25

Add ESP8266 to "architectures" in library.properties