SlashDevin/NeoGPS #26: Using Serial or Neoserial via NeoHWSerial rather than Serial1, SS or alternatives

xenek, 2017-05-01:
Hello SlashDevin, using an ESP8266 MCU...

I only have one hardware serial port. I am using it for the GPS. I have a physical switch to disable GPS & enable uploading of newly compiled code changes, or if the switch was removed I could also upload such firmware updates via a ESP8266 webserver. This system works perfectly - I have been using it with TinyGPS++, so I know my wiring and code around this is sound.

I do all my debug & monitoring using telnet thanks to: [https://github.com/JoaoLopesF/ESP8266-RemoteDebug-Telnet](url)

My issue is that I can't seem to get NeoGPS working showing satellites with fix when using Serial, even though the light on the GPS indicates it has a valid fix.

Not including all the wifi, lorawan, webserver & standard stuff It looks roughly like:

```
#include <NMEAGPS.h>
#include <RemoteDebug.h>

RemoteDebug Debug;
#define gpsPort Serial
NMEAGPS gps;
gps_fix fix;

void setup(void) {

Debug.printf("GPS port init \n");
gpsPort.begin(9600, SERIAL_8N1);

}

void loop(void) {

fix = gps.read();
Debug.printf("Satellites with fix: %u \n", fix.satellites);

}
```

I have tried using the great new tabular example Tabular.ino rather than just displaying the number of satellites as in the example above, but it seems to continue the tradition of using the only hardware serial port for debug messages & general output rather than allowing me to use it for the GPS... my modifications to it to send to the RemoteDebug library don't seem to result in any data either, though this might be due to the added complexity of your print feature. 

My thought is just that I could use NeoHWSerial with interrupt support on the sole hardware serial where a GPS is connected. Once I get just Serial working, I'll move onto using NeoHWSerial.

Would you have a tip?

xenek, 2017-05-01:
Well, I have a solution. Rather than stripping the code back to what I thought I understood, I left in 'while'. I now get a figure suggesting I have 7 satellites :)

```
while (gps.available( Serial )) {
fix = gps.read();
}
```

SlashDevin, 2017-05-01:
This would also work:

```
if (gps.available( Serial )) {
  fix = gps.read();
}
```
The `gps.available` method reads characters from `Serial`.  Once in a great while (after consuming hundreds of characters), a complete fix structure is available.  Then, and only then, you can read it.  :)

closed 2017-05-01 by SlashDevin