SlashDevin/NeoGPS #28: Example of a sketch with local configuration

andrzejtokarski, 2017-05-13:
Is it possible to configure the library without editing files downloaded by library manager? Is this library even usable with Arduino IDE?

SlashDevin, 2017-05-13:
> Is it possible to configure the library without editing files downloaded by library manager?

No.  The Arduino environment does not allow local includes to override Library includes.  I really wish it would, because that's what I'm used to, too.

You can either 

*  edit the config files in the Libraries/NeoGPS directory **or**
*  copy all the NeoGPS/src files into your sketch directory, preserving the directory structure.  Be sure to remove the Libraries/NeoGPS directory.

Old versions of NeoGPS required a copy in each sketch directory, not in the Libraries directory.  This also allowed different configs for each sketch.  But the current NeoGPS distribution is structured according to the Library Manager rules.

> Is this library even usable with Arduino IDE?

?

Of course it is usable with the Arduino IDE.  Are you having trouble using the library, when installed from the Library Manager?

Make sure you don't have two copies of NeoGPS files (one in Libraries/NeoGPS and one in your sketch directory).  This will cause strange errors, in either the build (compile or link) or the execution (crash, freeze, no data, weird data, etc.).

andrzejtokarski, 2017-05-13:
Thank you for the answer :) I see now that I'm having more issues with Arduino IDE then any of the code itself. 

It's a whole another question but do you know any reasons why the program would stop after getting a fix once ?

andrzejtokarski, 2017-05-13:
I'm reopening the issue because the compiler is stopping on:

> sketch\src\ublox\ubxGPS.cpp:1:25: fatal error: NMEAGPS_cfg.h: No such file or directory

NMEAGPS_cfg.h is present in the root sketch directory (as other files).

SlashDevin, 2017-05-14:
You do not need 3 includes in you sketch.  Just include "NMEAGPS.h".

Did you delete the Libraries\NeoGPS directory?  I haven't tried this directory structure for a while.

Unless you have multiple sketches with different configs, I would suggest leaving it in Libraries\NeoGPS directory.

andrzejtokarski, 2017-05-15:
The issue with this approach is that it's hard to track changes with Git when they are spread this way. Thank you for your time :)

closed 2017-05-15 by andrzejtokarski