SlashDevin/NeoGPS #29: Playing with extra functions

unocerobits, 2017-05-29:
Firstly, let me say you that this is a great library for GPS and UBX protocol!

I try to work with your ublox.ino pure UBX example, but I only need 2 new messages. I'm working with extra UBX functions (UBX only) in the M8U, this is a UDR with 3D IMU sensors, and a Teensy 3.2 board. 
This new messages are NAV-ODO (0x01 0x09) and HNR-PVT(x028 0x00) . 

I add this, following your DOP code as reference, in `ubxGPS.cpp` :
```
bool ubloxGPS::parseNavODO   ( uint8_t chr )
{
  bool ok = true;

//if (chrCount == 0) Serial.print( F( "dop ") );
  #ifdef UBLOX_PARSE_ODO
    switch (chrCount) {
      case 4: case 5: case 6: case 7:
        ok = parseTOW( chr );
        break;

      #ifdef GPS_FIX_ODO_DISTANCE
        case 8: case 9: case 10: case 11:
          NMEAGPS_INVALIDATE( distance );      
          ((uint32_t *)&m_fix.distance) [ chrCount-6] = chr;
          m_fix.valid.distance = true;
          break;
        
      #endif

      #ifdef GPS_FIX_ODO_TOTAL_DISTANCE
        case 12: case 13: case 14: case 15:  
          NMEAGPS_INVALIDATE( total_distance );      
          ((uint32_t *)&m_fix.total_distance) [ chrCount-12] = chr;
          m_fix.valid.total_distance = true;
          break;
      #endif

      #ifdef GPS_FIX_ODO_DISTANCE_STD
        case 16: case 17: case 18: case 19:
          NMEAGPS_INVALIDATE( distanceSTD );
          ((uint32_t *)&m_fix.distanceSTD) [ chrCount-16] = chr;
          m_fix.valid.distanceSTD = true;
          break;
      #endif

    }
  #endif

  return ok;

} // parseNavODO 
```
And add `UBX_NAV_ODO` case in `ubloxGPS::parseField` function and `bool parseNavODO ( uint8_t chr );' in 'ubx.GPS.h.
`
I checked that It parses ODO message (0x01 0x09) correctly, but I don't know how receive into `<Streamers.h>` or my sketch directly (as  'if (fix.valid.distance) Serial.print( fix.distance / 1000 );'

Would be great if you can explain/example us how can I try to check this data and how HNR-PVT could be implemented.

Thanks again,
Best regards!

SlashDevin, 2017-05-29:
Very nice!

Did you add the new fix pieces to `GPSfix.h`?  The above code seems to show that they are GPS_FIX options (in `GPSfix_cfg.h`) and are part of `m_fix` (from`GPSfix.h`).

If you want the example programs (or your modded versions) to print them from `Streamers.h`, it's just a matter of picking where you want them to appear in the CSV line.  You will need `#ifdef` sections in the header string (line 24+).

You also need a new `#ifdef` section in `operator <<` (line 134+), in both the FLOAT and NOT FLOAT sections.  Be careful not to use `float` in the NOT FLOAT section.

In your sketch, you should be able to use those new members if they are declared `GPSfix.h` and enabled in `GPSfix_cfg.h`.

unocerobits, 2017-05-29:
ouuch!. nop, Thanks!!

I have wrote this code in GPSfix.h:
```
#ifdef GPS_FIX_ODO_DISTANCE
    uint32_t           distance;
  #endif

  #ifdef GPS_FIX_ODO_TOTAL_DISTANCE
    uint32_t           total_distance;
  #endif

  #ifdef GPS_FIX_ODO_DISTANCE_STD
    uint32_t           distanceSTD;
  #endif
```
And this in GPSfix_cfg.h:
```
#define GPS_FIX_ODO_DISTANCE
#define GPS_FIX_ODO_DISTANCE_STD
#define GPS_FIX_ODO_TOTAL_DISTANCE
```
but compilation fails with: `Streamers.cpp:261:21: error: 'const struct gps_fix::valid_t has no member named 'distance'  if (fix.valid.distance)' `

In Streamers.h, as I put ODO messages as FIX, I understand that I don't need add nothing in this (¿?) 

In Streamers.cpp, I add this:
 - In `gps_fix_header[]`:
```
#if defined(GPS_FIX_ODO_DISTANCE)
    "ODO,"
  #endif
  #if defined(GPS_FIX_ODO_TOTAL_DISTANCE)
    "D. Total,"
  #endif
  #if defined(GPS_FIX_ODO_DISTANCE_STD)
    "D. STD,"
  #endif
```
And 'operator <<' lines:
 - In `USE_FLOAT`
```
 #ifdef GPS_FIX_ODO_DISTANCE
      if (fix.valid.distance)
        outs.print( (fix.distance * 0.001), 3 );
      outs << ',';
    #endif
```
 - And the fails says here: In `not USE_FLOAT`:
```
#ifdef GPS_FIX_ODO_DISTANCE
      if (fix.valid.distance)
        outs << fix.distance;
      outs << ',';
    #endif
```
If I change the name in `GPSfix_cfg.h` and `GPSfix.h` compile OK, but nothing displays, only `acquiring... `

And curiosity question, I have worked with some libraries but this is advanced library that I had touched, which IDE have you worked with? Intellisense (or similar) would be essential I sense. 😅
Thanks again for your time.

SlashDevin, 2017-05-29:
>   I don't need to add anything in `Streamers.h` (¿?)

Correct.  The modifications to `Streamers.cpp` look correct.

> `gps_fix::valid_t has no member named 'distance' if (fix.valid.distance)`

There is a structure in `GPSfix.h` (`valid_t` on line 248) that contains all the valid flags.  You need to add one new `bool` for each piece you want to mark as valid or invalid.  Use the same names you used in `parseNavODO`.

> If I change the name

What name?

You may need to review which UBX sentence arrives last in each interval, in `ubx_cfg.h`.  Your new sentences can affect this choice.

> nothing displays, only `acquiring...`

If you look at `ublox.ino`, you'll see that it needs to receive a few sentences before it will ask for locations (line 80).  If your sketch is based off of that, make sure you are still requesting those sentences.  That message means that it is waiting for a STATUS message.  Make sure `UBLOX_PARSE_STATUS` is enabled in `ubx_cfg.h`.  If it is, you might check the return value at line 130:

```
        if (requestNavStatus) {
          // Turn on the UBX status message
          if (!enable_msg( ublox::UBX_NAV, ublox::UBX_NAV_STATUS ))
            DEBUG_PORT.println( F("enable STATUS failed!") );
        }
```

unocerobits, 2017-05-29:
Great SlashDevin! 

I had wroten NAV-ODO and HNR-PVT, your library works great. I obtain HNR-PVT messages at 20Hz and NAV-ODO at 10Hz at 115200bps, great performance!

I had some problems, mainly, with ` #define UBX_LAST_MSG_CLASS_IN_INTERVAL   ublox::UBX_HNR`. If last message is `UBX_NAV` goes blank data (but received ok). Also I have used the same GPSfix defines with the same instructions, per example, `GPS_FIX_DATE` from `NAV-TIMEUTC` is invalidated by `GPS_FIX_DATE` from `HNR-PVT`. With this I have more info with less messages ( I think).

Finally and following your first comment, I need to redirect Streamers `trace_all()` function into my sketch (I will show distance in SSD1306 oled display), but Streamers.h does not let me return nothing. I see some example to call data from sketch, that I have modified, for example inside `loop()`, if I want to show distance (distance travelled from `NAV-ODO`):
```
while (gps_port.available()) {
    if (gps.decode( gps_port.read() ) == NMEAGPS::DECODE_COMPLETED) { // <-- Is HNR and ODO  inside NMEAGPS class?
      if (gps.fix().valid.distance)
        distance = gps.fix().distance / 1000;
}
```
Is correct this method?
Thanks again!

SlashDevin, 2017-05-30:
> I had some problems, mainly, with #define UBX_LAST_MSG_CLASS_IN_INTERVAL ublox::UBX_HNR.

ok.

> If last message is UBX_NAV goes blank data (but received ok).

I don't understand "goes blank data (but received ok)."

> Also I have used the same GPSfix defines with the same instructions, per example, GPS_FIX_DATE from NAV-TIMEUTC is invalidated by GPS_FIX_DATE from HNR-PVT.

Yes, any part of a fix can be invalidated/validated multiple times.  This happens when the same fix member is in two different messages.

> With this I have more info with less messages ( I think).

I'm not sure.

>  I need to redirect Streamers trace_all() function into my sketch (I will show distance in SSD1306 oled display)

Not usually.  The Streamers functions are just for displaying the [CSV format](https://en.wikipedia.org/wiki/Comma-separated_values) lines:

```
3,2016-05-24 01:21:29.00,472852332,85652650,,138,,,1,0,66,
```

Is that what you want to see on the display?

If not, just print the pieces you want.  Don't use the Streamers functions.  `Tabular.ino`, `NMEAloc.ino` and `NMEAsimple.ino` do not use Streamers functions.  All of the pieces in the fix (see [Data Model](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Data%20Model.md) page) can be printed to the display.  Just set the cursor position and print one piece.

unocerobits, 2017-05-30:
Thanks for your info! 

In the display, I show distance travelled, time and speed (3D), I try with this in loop():
```
while (gps.available( gps_port )){
    //trace_all( DEBUG_PORT, gps, gps.read() );
    gps_fix fix = gps.read();
    if (fix.valid.distance){
      DEBUG_PORT.print(gps.fix().distance); 
      DEBUG_PORT.print(gps.fix().speed3D);
      DEBUG_PORT.println(gps.fix().timeNano);
    }
```
My last question has relation with "blank screen" (I called "blank screen" from my ignorance, sorry) I had disabled all messages except NAV-ODO and HNR-PVT, and your example sketch needs NAV-STA to start (I think) sending messages. The last question is how can I disable/change NAV-STA by another (example NAV-ODO) to start? (I don't need NAV-STA because HNR-PVT send me the status values)

Thanks again for your answers!

SlashDevin, 2017-05-30:
Do not use `gps.fix()`, because it changes as characters are received.  You should just use the `fix` structure returned from `gps.read()`:

```
while (gps.available( gps_port )){
    //trace_all( DEBUG_PORT, gps, gps.read() );
    gps_fix fix = gps.read();
    if (fix.valid.distance){
      DEBUG_PORT.print(fix.distance); 
      DEBUG_PORT.print(fix.speed3D);
      DEBUG_PORT.println(fix.timeNano);
    }
```

> I had disabled all messages except NAV-ODO and HNR-PVT, and your example sketch needs NAV-STA to start (I think) sending messages. The last question is how can I disable/change NAV-STA by another (example NAV-ODO) to start? (I don't need NAV-STA because HNR-PVT send me the status values)

The method `get_status` requests the NAV-STA message here:

```
    void get_status()
    {
      static bool acquiring = false;

      if (fix().status == gps_fix::STATUS_NONE) {
            ...
        if (requestNavStatus)
          // Turn on the UBX status message
          enable_msg( ublox::UBX_NAV, ublox::UBX_NAV_STATUS );
```

When it finally receives that message, it doesn't turn it off, but you could:

```
      } else {
        if (acquiring)
          DEBUG_PORT << '\n';
        DEBUG_PORT << F("Acquired status: ") << (uint8_t) fix().status << '\n';
        disable_msg( ublox::UBX_NAV, ublox::UBX_NAV_STATUS );
```

Then it asks for TIMEGPS and TIMEUTC, and then calls `start_running`.  That is where you would enable the HNR-PVT.  I don't think I would use the HNR-PVT to get the status during start-up (`get_status`).  You could, but you have to be careful not to use the time members without knowing leap seconds and UTC-to-TOW (time of week).

I just realized that ublox.ino uses `fix()` during start-up instead of reading a fix structure.  This is not safe, so I need to update that start-up sequence: `get_status`, `get_leap_seconds` and `get_utc` need to change.

unocerobits, 2017-05-31:
Thanks again for your excellent work and answer! Works perfect!

unocerobits, 2017-06-02:
Hi again, sorry by reopen this thread.

All works perfectly, but when I saw Distance (from NAV-ODO) values, only goes to 255 (uint8_t). In the code I did put:
```
case 8: case 9: case 10: case 11:
          NMEAGPS_INVALIDATE( distance );      
          ((uint32_t *)&m_fix.distance) [ chrCount-8] = chr;
          m_fix.valid.distance = true;
          break;
```
And the sketch I have declared:
```
uint32_t distance;
[..]
while (gps.available( gps_port )) {
    gps_fix fix = gps.read();
    if (fix.valid.distance) {
      distance = fix.distance;
#ifdef DEBUG_ON
      DEBUG_PORT.print( F("Distancia: ") );
      DEBUG_PORT.println(distance);
#endif
```
And GPSfix.h
```
#ifdef GPS_FIX_ODO_DISTANCE
    uint32_t           distance;
  #endif
[...]
 #ifdef GPS_FIX_ODO_DISTANCE
      bool distance NEOGPS_BF(1);
    #endif
[...]
#ifdef GPS_FIX_ODO_DISTANCE
      distance = 0;
    #endif
[...]
#ifdef GPS_FIX_ODO_DISTANCE
      if (r.valid.distance)
        distance = r.distance;
    #endif
```

I have not problem with show it in meters, I understand that I don't need transformate this values. But I don't know why distance values seem `uint8_t`.

SlashDevin, 2017-06-02:
Ah, I just noticed how you were accumulating 32-bit values:

```
      #ifdef GPS_FIX_ODO_TOTAL_DISTANCE
        case 12: case 13: case 14: case 15:  
          NMEAGPS_INVALIDATE( total_distance );      
          ((uint32_t *)&m_fix.total_distance) [ chrCount-12] = chr;
          m_fix.valid.total_distance = true;
          break;
      #endif
```

That invalidates and validates on every character.  Instead, you need to invalidate on the first character (chrCount == 12) and validate on the last character (chrCount == 15):

```
      #ifdef GPS_FIX_ODO_TOTAL_DISTANCE
        case 12:
          NMEAGPS_INVALIDATE( total_distance );      
        case 13: case 14: case 15:  
          ((uint32_t *)&m_fix.total_distance) [ chrCount-12] = chr;
          if (chrCount == 15)
            m_fix.valid.total_distance = true;
          break;
      #endif
```
Do this for all 3 multi-byte pieces.  This **does not** explain why `total_distance` is always <= 255.  This could be a memory corruption.  You don't have any \*_cfg files in the sketch directory, do you?

Are you using the `nav_dop_t` structure I defined in `ubxGPS.h`?

unocerobits, 2017-06-02:
Yes, I am using your `nav_odo_t` structure from `ubxGPS.h`.

I think that my problem can be related to using  `((uint32_t *)&m_fix.total_distance)` because I have been studying your others parses, and I don't understand well, by example, in speed from parseNavVelNED, you assign `uint8_t `to `uint32_t `gSpeed (from Ublox M8 Protocol Manual)
```
// Temporarily store the 32-bit cm/s in the spd member
((uint8_t *)&m_fix.spd) [ chrCount-20 ] = chr;
```
In my case, the variable `total_distance `increase (by GPS motion) to 255 and then start in 0 continuously.

SlashDevin, 2017-06-03:
Yes, that's it.

```
((uint8_t *)&m_fix.spd) [ chrCount-20 ] = chr;
```

That assignment statement:

1.  Takes the address of the member (a 32-bit int or whatever) with the `&` operator ("address-of").  That's the RAM location of the bytes for that member.  The `&` operator returns a pointer of type `uint32_t *`, the *address* of a uint32_t.

2.  The `(uint8_t *)` converts (i.e. *casts*) that pointer to a pointer-to-byte, not a long.

3.  The `[ n ]` array index uses the pointer as the starting address of an array, and returns the **nth** element of that array.  It steps through the array by the number of bytes in each element.  Because the pointer is "declared" to be an address of `uint8_t`, each element is one byte.  The address of each successive element increase by 1.

4.  The character value from `chr` is assigned to that element.

The member is treated as a byte array, so that it can put *one byte at a time* into a specific place in the member (a 32-bit int of 4 bytes). 

Since you cast it to a pointer-to-uint32_t, the `[ ]` operator steps by 4 bytes at a time through the array.  The first character goes in the right place, but the other three are stepping on something else.  The assignment also copies 4 bytes at a time (instead of one) by extending the `chr` from 8-bits to 32-bits.

Change yours to `uint8_t *` and I'll bet it works a lot better.  Sorry I didn't notice.

unocerobits, 2017-06-03:
Yes sir, `uint_8` change works perfectly!
Thanks again! I don't know if there is any possibility to thank you the time that you have dedicated to my help, you don't doubt to say me it

SlashDevin, 2017-06-03:
Glad to help

closed 2017-06-03 by SlashDevin