SlashDevin/NeoGPS #30: Power management?

laptopdude90, 2017-06-29:
Is there any way I can use this library to shift the GPS into low power mode?

SlashDevin, 2017-06-29:
Different brands may have configuration commands you can send.  NeoGPS has a method to send NMEA sentences to a device.  It adds the '$' and the checksum at the end:

```
    gps.send_P( &gpsPort, F("PMTK432,234,5,67") );
```

Some brands require you to send binary commands, not NMEA text commands.

What brand/model of GPS device are you using?

laptopdude90, 2017-06-30:
I'm using the Neo M8Q. I'm decently sure it doesn't need binary commands; NMEA should be sufficient based on what I've read of the datasheet.

SlashDevin, 2017-06-30:
> NMEA should be sufficient based on what I've read of the datasheet.
Which NMEA command did you want to send?

laptopdude90, 2017-06-30:
I think I've got it from here, but thank you :) I'll let you know if I get stuck.

closed 2017-07-03 by SlashDevin

laptopdude90, 2017-07-04:
Okay, I'm confused.

I'm pretty sure this is the command I want: UBX-RXM-PMREQ

I'm not sure how to give it arguments or even what arguments to use. Here's the documentation: https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_(UBX-13003221)_Public.pdf

All I want to do is turn off the GPS for a set amount of time. I know this is less to do with your library and more to do with me having difficulty reading the documentation, but I'm not sure where else to turn.

SlashDevin, 2017-07-05:
Sure, I assume you have read Section 12, "Power Management".  You can manually control the power modes, based on your own criteria: time, location, etc.  You would turn it on or off for an "infinite" amount of time (duration == 0).  You can also tell the receiver to do something based on its internal time (a duration > 0 and < 12 days).

From that, you just need to determine the bytes you should send.  There is an example in NeoGPS for setting the reporting rate and baud rate (look [here](https://github.com/SlashDevin/NeoGPS/blob/master/examples/ubloxRate/ubloxRate.ino#L93)).  Your byte array should look something like this:

```
const unsigned char ubxPMREQ[] PROGMEM = 
  { 0x02, 0x41, 0x08,0x00, 0,0,0,0, 0x02 };
```
Bytes at ubxPMREQ[4..7] are the duration, defaulted to zero above.  Set them to whatever you want (Little Endian 32-bit milliseconds), and call `sendUBX`.  This is a FLASH array, so you can't change it at run-time.  Declare several different arrays if you want different times.

charlienicholsonr, 2023-11-12:
I hate to wake up a closed issue, but I was wondering how I could do this with the newer MAX-M10S modules. I have read the interface description, but I'm a little confused about how to get all that information into an array and send it over I2C using NeoGPS. So far, [using the interface manual](https://content.u-blox.com/sites/default/files/u-blox-M10-SPG-5.10_InterfaceDescription_UBX-21035062.pdf), I have gotten this array: 
```
const unsigned char UBX-RXM-PMREQ[] PROGMEM = {0x5b, 0x62, 0x02, 0x41, 16, 5000, 0, 0, 0, 1, 0, 0};
```
My reasoning for this: 0x5b, 0x62 (headers), 0x02 (class), 0x41 (ID), 16 (length), 5000 (sleep time in ms), 0 (no backup), 0 (no minimum power consumption), 0 (no UART wakeup), 1 (extint0 wakeup), 0 (no extint1 wakeup), 0 (no SPI wakeup).

I don't think I did it right, especially because I don't know how to do checksums with u-blox. But I also don't know how to send the packet over I2C with this library. Any help would be greatly appreciated! I apologize in advance for my lack of knowledge.