SlashDevin/NeoGPS #32: Page blanc without data or errors !

AEK-BKF, 2017-07-11:
Hi man,
Thank you so much for this nice package !
I've imported your library and tested it on Arduino Uno (pinsRx =0, pinxTx = 1).

Please what's the mistake ?

Here my program :
```
#include <NMEAGPS.h>
  /*
    This program sketch obtain and print the lati,logi,speed,date and time
    It requires the use of SoftwareSerial, and assumes that you have a
    9600-baud serial GPS device hooked up on pins 4(rx) and 3(tx).
   */
#include <NeoSWSerial.h>
static const int RXPin = 0, TXPin = 1;
NeoSWSerial gpsPort(RXPin, TXPin);
static const uint32_t GPSBaud = 9600;

NMEAGPS gps;
gps_fix fix;
uint8_t fixCount = 0;

void setup()
{
  Serial.begin(9600);
  gpsPort.begin(GPSBaud);

  Serial.println(F("GPS LOADING....."));
  Serial.println(F("Obtain and print lati,logi,speed,date and time"));
  Serial.println(F("Testing by : "));
  Serial.println(F("Billa"));
  Serial.println();
}

void loop()
{
  while (gps.available( gpsPort )) {
    fix = gps.read();

    // Once every 5 seconds...    
    if (++fixCount >= 5) {
      displayInfo();
      fixCount = 0;
    }
  }

  if ((gps.statistics.chars < 10) && (millis() > 5000)) {
     Serial.println( F("No GPS detected: check wiring.") );
     while(true);
  }
}

void displayInfo()
{
  Serial.print(F("Location: ")); 
  if (fix.valid.location) {
    Serial.print( fix.latitude(), 5 );
    Serial.print( ',' );
    Serial.print( fix.longitude(), 5 );
  } else {
    Serial.print(F("INVALID"));
  }

  Serial.print(F("  Speed: ")); 
  if (fix.valid.speed) {
    Serial.print(fix.speed_kph());
    Serial.print(F(" KMPH "));
  } else {
    Serial.print(F("INVALID"));
  }

  // Shift the date/time to local time
  NeoGPS::clock_t localSeconds;
  NeoGPS::time_t  localTime;
  if (fix.valid.date && fix.valid.time) {
    using namespace NeoGPS; // save a little typing below...

    localSeconds = (clock_t) fix.dateTime; // convert structure to a second count
    localSeconds += 5 * SECONDS_PER_HOUR + 30 * SECONDS_PER_MINUTE; // shift timezone
    localTime = localSeconds;              // convert back to a structure
  }

  Serial.print(F("  Date : "));
  if (fix.valid.date) {
    Serial.print(fix.dateTime.month);
    Serial.print('/');
    Serial.print(fix.dateTime.date);
    Serial.print('/');
    Serial.print(fix.dateTime.year);
  } else {
    Serial.print(F("INVALID"));
  }

  Serial.print(F("  Time : "));
  if (fix.valid.time) {
    Serial.print(localTime.hours);
    Serial.print(':');
    if (localTime.minutes < 10) Serial.print('0');
    Serial.print(localTime.minutes);
    Serial.print(':');
    if (localTime.seconds < 10) Serial.print(F("0"));
    Serial.print(localTime.seconds);
  } else {
    Serial.print(F("INVALID"));
  }

  Serial.println();
}

```

AEK-BKF, 2017-07-11:
I solved it , I put the right baud rate 38400 ^^' 
Thanks :))))

SlashDevin, 2017-07-11:
Ok, the GPS device baud rate is 38400, but you have a problem here:

```
#include <NeoSWSerial.h>
static const int RXPin = 0, TXPin = 1;
NeoSWSerial gpsPort(RXPin, TXPin);
```
Since you are using an UNO, pins 0 and 1 are the `HardwareSerial` pins for the `Serial` object.  It's ok to use those pins for the GPS, but don't use `NeoSWSerial` to control them.  You can just do this:

```
#define gpsPort Serial
```

... which means you don't need to call `begin` for both `Serial` and `gpsPort`.  They are the same port.  `setup` could just be this:

```
void setup()
{
  Serial.begin( GPSBaud );
  //gpsPort.begin( GPSBaud );  // not needed if gpsPort *is* Serial

  Serial.println(F("GPS LOADING.....\n"));
}
```
The `Serial` object is much more efficient and reliable than a `NeoSWSerial` object.

closed 2017-07-16 by SlashDevin