SlashDevin/NeoGPS #34: $GPTXT sentence

Sigafoos, 2017-07-20:
(before I begin, I know the answer is almost certainly going to be "write a [new parser](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Extending.md) for it", and I probably will, but)

I got a ublox neo 6m and ran NMEAdiagnostic.ino: the ublox is transmitting on 9600, and in the test I see the sentences coming through, but it's `$GPTXT`, which isn't one of the default message types, so the code continued to scan.

Since this is an out-of-the-box thing I got from Amazon I was surprised that there was something that wasn't supported (since ublox is specifically). Like I said, I know I can make NeoGPS accept anything, but has it already been done by someone? (not that I could tell from searching) If I wrote it, would you consider a PR to add it? Or does this mean I got a malfunctioning ublox?

jamesmayes, 2017-07-20:
Hi Dan,

Did you see the ublox forum post suggesting the possibility of a tx/rx
short?

James

Sigafoos, 2017-07-20:
I didn't! Damn, I thought I had done my research before posting, too. I'll try to look into it tonight.

SlashDevin, 2017-07-23:
Yes, I have occasionally seen the GPTXT come through, but I've always ignored them.  I've never seen ONLY the GPTXT, though.

Regarding parsing them, I think multiple GPTXTs could be received in each update interval, so these sentences look more like an input stream.  Questions of buffering, callbacks and/or virtuals arise.  Do results go in a `gps_fix`, or are the contents delivered some other way?

It's an interesting problem, though.  I'm always open to suggestions, and would be happy to discuss approaches.

closed 2017-08-10 by SlashDevin