SlashDevin/NeoGPS #35: Having trouble with latitudeDMS.NS

laptopdude90, 2017-07-22:
This is almost definitely me overlooking something really dumb.

char ffNS = (fix.latitudeDMS.NS);

I get "error: cannot convert 'DMS_t::NS' from type 'char (DMS_t::)() const' to type 'char'"

I also get an error if I try using it in a memcpy:

memcpy(fix.latitudeDMS.NS, &NMEALat[7], 1);

"error: invalid use of non-static member function"

Sorry for the probably noobish question!

laptopdude90, 2017-07-22:
I was at it for hours and I just realized what I did.

laptopdude90, 2017-07-22:
()

SlashDevin, 2017-07-23:
Dang parens!

```
char ffNS = fix.latitudeDMS.NS();
```
Glad to help!  XD

laptopdude90, 2017-07-23:
I have a new problem!

`gps_fix fix; // = gps.read();
            int attempts = 0; //# of attempts trying to get a lock
            while(!fix.valid.location)
            {
                attempts++;

                while(gps.available(Serial1))
                {
                    fix = gps.read();
                }
                delay(250);

                if(attempts > 3)
                {
                    //Wait for a little bit, then reset the attempts and keep trying.
                    unsigned long lastAttemptTimems = getRealMillis();
                    digitalWrite(GPS_PWR, LOW);
                    while((unsigned long)(getRealMillis() - lastAttemptTimems + 32000) <= settings.NoLockDelayms)
                    {
                        deepSleep();
                    }
                    digitalWrite(GPS_PWR, HIGH);
                    deepSleep();
                    attempts = 0;
                }
            }`

This should attempt to get a lock a few times. If it fails, it should sleep for settings.NoLockDelayms amount of time. The problem is it always fails, even though I see the light blinking on the GPS.

SlashDevin, 2017-07-23:
If no GPS characters are available, or not enough have been received to emit a `fix`, the `while` loop exits immediately.  Then you wait 250ms.  Up to 250 characters could be received during this time (@9600 baud), but the input buffer only has room for 64.  So it drops the remaining characters, and NeoGPS never gets to parse a complete batch of sentences.

In general, **NEVER** use `delay` in a GPS program.  Instead, call `gps.available(Serial1)` all the time.  This is called a "non-blocking" approach.  Then you can use the fixes as a 1-second clock:

```
static int attempts = 0;

void loop()
{
  while(gps.available(Serial1)) {
    fix = gps.read();

    if (fix.valid.location) {
      attempts = 0;
      // Do something with the fix here?  Maybe nothing, if `fix` is used elsewhere.

    } else { // no valid location yet
      attempts++;
      if (attempts > 3) {
        waitForALittleBit();
        attempts = 0;
      }
    }
  }
}

void waitForALittleBit()
{
  unsigned long lastAttemptTimems = getRealMillis();
  digitalWrite(GPS_PWR, LOW);
  while((unsigned long)(getRealMillis() - lastAttemptTimems + 32000) <= settings.NoLockDelayms)
  {
      deepSleep();
  }
  digitalWrite(GPS_PWR, HIGH);
  deepSleep();
}
```

9600, 2017-07-23:
You called?

SlashDevin, 2017-07-23:
LOL

laptopdude90, 2017-07-23:
Ah, that would do it. I had a feeling I was missing something.

Thank you very much!

closed 2017-07-31 by SlashDevin