SlashDevin/NeoGPS #39: fix.lat_err(), fix.lon_err() and fix.alt_err() -

githubtd01, 2017-08-04:
I have the impression that fix.lon_err_cm and fix.alt_err_cm / fix.lon_err() and fix.alt_err() are swaped by the parser.

SlashDevin, 2017-08-06:
Well, according to [this](http://www.catb.org/gpsd/NMEA.html#_gst_gps_pseudorange_noise_statistics) and the ublox NEO-6 protocol spec, the GST sentence sends them as fields 6 (lat), 7 (lon) and 8 (alt).  The parser handles those fields in the same order:

```
bool NMEAGPS::parseGST( char chr )
{
  #ifdef NMEAGPS_PARSE_GST
    switch (fieldIndex) {
      case 1: return parseTime( chr );
      case 6: return parse_lat_err( chr );
      case 7: return parse_lon_err( chr );
      case 8: return parse_alt_err( chr );
    }
  #endif

  return true;

} // parseGST
```
... and those subroutines appear to set the corresponding members of `gps_fix`.

Why do you think they are swapped?

closed 2017-08-10 by SlashDevin

githubtd01, 2017-08-10:
Hi, 
I'm sorry for the late reply. I got very strange values, the alt value was in the same range as the lat value and the lon value was 5 times higher. I'm using a Neo m8n with the latest firmware. I did not manage to find the part of the code where the errors are parsed.

Thank you for checking and all your work!