SlashDevin/NeoGPS #41: Implementing NeoGPS for Arduino Mega

stevenette, 2017-08-16:
Hello, I have a Garmin LVC 18x GPS that I got working with an arduino Uno with an SD shield. Unfortunately, the Uno was not large enough for the program I am trying to eventually write, so I switched everything to a Mega 2560. The Mega however does not like software serial, which, is what TinyGPSPlus uses. So many forwarded me to NeoGPS. That being said, I have fixed many issues and errors that have popped up, and eventually upgraded the Arduino IDE to get all of them to finally go away. Now, my issue is not an error popping up, but nothing popping up. 

I am using the basic NMEA.ino example. 
My GPS is 4800 Baud rate (Which I have changed around line 145). 
I have switched the RX and TX ports endlessly on 18 and 19 in case I got them mixed up

I tried the original GPSport.h, as well as commented out everything except 
#define gpsPort Serial1
#define GPS_PORT_NAME "Serial1"
#define DEBUG_PORT Serial

And still all I get out of the serial window is
NMEA.INO: started
  fix object size = 31
  gps object size = 84
Looking for GPS device on Serial1
GPS quiet time is assumed to begin after a RMC sentence is received.
  You should confirm this with NMEAorder.ino
Status,UTC Date/Time,Lat,Lon,Hdg,Spd,Alt,Sats,Rx ok,Rx err,Rx chars,

When running NMEAorder.ino, I get
No valid sentences, but characters are being received.
Check baud rate or device protocol configuration.

I know the GPS is working because I have just re-tested it on the Uno and it spits out NMEA sentences like $--GGA. The GPS has been sitting in the same spot, so it's not still warming up. I also tested it with NMEADiagnostics and it said the same thing "No valid sentences, but characters are being received" With no conclusion on baud rate (Specs say 4800 which works on Uno)

When I run a while(serial1.available) loop I get A LOT of numbers but no ascii like below
155
78
159
141
167
157
163
157
167
This goes on ad naseum. 

I have no idea where else to go. Thank you.

SlashDevin, 2017-08-16:
>I am thinking I need to invert the signal logic? That is what I did with SoftwareSerial(rx,tx,1) but I have not found a way to do this on the Mega 2560 without a 232 shifter? Does this sound like it may be the problem?

Yes, if you had to invert the logic for `SoftwareSerial`, you would need to invert the signal before connecting it to Serial1 RX (pin 19).  There is no way to invert the logic on a `HardwareSerial` port (i.e., `Serial`, `Serial1`, `Serial2` or `Serial3`).  You could do this with a PNP transistor and a few resistors.

close2, 2017-08-19:
If you have 2 pins (let's call them PIN_A and PIN_B) free on your Mega 2560 a quick _workaround_ would be to simply output everything form PIN_A inverted on PIN_B (using an interrupt function).

It might even be possible to put the Serial_input pin into output mode and directly write to that pin.  In that case all you need is 1 other free pin.  (Please tell us, if this works :)

closed 2017-08-23 by SlashDevin