SlashDevin/NeoGPS #43: Using NeoGPSTabular example with Teensy 3.x

mjs513, 2017-08-29:
While trying to run the tabular example in the library I ran into the following error: 
`Tabular: In function 'const __FlashStringHelper* compassDir(uint16_t)':
Tabular:200: error: 'pgm_read_ptr' was not declared in this scope
   return (const __FlashStringHelper *) pgm_read_ptr( &dirStrings[ dir ] );`

I am using Arduino IDE 1.8.4 with the latest version of the Teensyduino (1.39 beta 1) and compiling for a Teensy 3.5.

Do not know how to fix this one.

Thanks
Mike

mjs513, 2017-08-30:
Yes I do.

mjs513, 2017-08-30:
It is now fixed in the release of Teensyduino and works fine.  So I am going to go ahead and close this now

closed 2017-08-30 by mjs513

SlashDevin, 2017-08-30:
Great, thanks!  I had seen a similar problem in NMEAGPS.cpp, and added Teensy-specific code to work around the issue.  Maybe I can take that out now.

mjs513, 2017-08-30:
Maybe.  Not sure what version of Teensyduino you are using but it has been evolving.   The things that were missing from pgmspace was pgm_ptr macros.  Paul added the near and far.

mjs513, 2017-08-31:
Hi SlashDevlin

I just commented out all the AVR ifdefs and everything from #else down to #endif in nemagps.cpp and reran the tabular.ino example.  It compiled and ran fine using the ide combo that I mentioned earlier.  Check this thread https://forum.pjrc.com/threads/46089-Teensyduino-1-39-Beta-1 for latest info on the releases.

Mike