SlashDevin/NeoGPS #44: Can't Get Longitude and Latitude Data from Neo-6m Module

aarush22, 2017-08-30:
Hi,
I am trying to use Neo-6m module with Arduino Uno, but unfortunately I have not been able to get the data from the module. I have attached a screen-shot of the raw data recieved with NMEA.ino example.
I am using AltSoftSerial in the GPSport.h, for communication and connected my GPS TX line to pin 8 and GPS RX line to pin 9 of arduino. I have been through almost all the GPS libraries availaible online for arduino but none seem to work. Can anyone help me?
[image]

SlashDevin, 2017-08-30:
It looks like the GPS is connected correctly, because NMEA.ino is printing once per second.  As it says on the [Troubleshooting](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Troubleshooting.md#poor-reception) page, the GPS device may not be receiving any information from the satellites.  Please read through that section.  I would guess that you are inside, or that your antenna is not connected or is defective.

Did you run NMEAorder.ino to confirm that the LAST_SENTENCE_IN_INTERVAL is really the GLL sentence?

Are you using the default configuration files?

To include program output, it is better to select the text in the Serial Monitor window (with the mouse), then copy it (ctrl-C0), and paste it into a comment here.  Put three grave accents ("\`\`\`") before the text and after it to make it appear like this:

```
0,,,,,,0,16,0,438,
0,,,,,,0,16,0,438,
0,,,,,,0,16,0,438,
0,,,,,,0,16,0,438,
```

This is much better than pasting an image of the text, because I can barely read the image.  ;)  The same goes for posting code.  It also allows me to select the text and paste it into my own editors.

defogprog, 2017-08-31:
If you're trying to receive GPS data indoors, then try to place receiver as close as possible to the window or go outside. GPS signal usually is too weak when receiver is inside the house.

aarush22, 2017-08-31:
Yes I checked with NMEAorder.ino, the last sentence is GLL, here's the output : 

'''NMEAorder.INO: started
fix object size = 31
NMEAGPS object size = 185
Looking for GPS device on Serial
..........
Sentence order in each 1-second interval:
  RMC
  VTG
  GGA
  GSA
  GLL

SUCCESS: LAST_SENTENCE_IN_INTERVAL is correctly set to NMEAGPS::NMEA_GLL'''

I am using the default config files, the only thing I changed is the serial port for communication to AltSoftSerial in GPSport.h. 
I read the Troubleshooting page and I uncommented the lines mentioned in NMEAGPS_cfg.h to view satellite tracking info, and I went outside in a open Area so that the GPS has an unobstructed view of the sky, still I am not able to receive any data. Occasionally the the output shows a number in the 'satellites being tracked' field but its only an impulse once in a while without any other info. Here's a sample output

'''0,,,,,,,0,[],625,1,18049,
0,,,,,,,0,[],630,1,18193,
0,,,,,,,0,[8 0/0@29,],636,1,18363,
0,,,,,,,0,[],641,1,18507,
0,,,,,,,0,[],646,1,18651,
0,,,,,,,0,[],651,1,18795,
0,,,,,,,0,[],656,1,18939,
0,,,,,,,0,[],661,1,19083,
0,,,,,,,0,[],666,1,19227,
0,,,,,,,0,[12 0/0@28,],672,1,19397,
0,,,,,,,0,[],677,1,19541,
0,,,,,,,0,[],682,1,19685,
'''

I might have a faulty antenna, as you said, but since this is a new module which I got only a few days ago, the chances are less.
Anything else I can try?

P.S.-Thanks for the advice about Styling the code 👍 )

mjs513, 2017-08-31:
If you are doing your tests indoors it can take up to 15 minutes to acquire the satellites and start getting data.  Neo-6m was the worst performer in time to acquire, Neo-8m and Neo-8u were the best especially since I use the assist now offline to help in the ublox center. Just thought I'd mention this since I saw the screen shot just starting with the data.  Have you let it warm up enough?

SlashDevin, 2017-08-31:
Another possibility is insufficient or noisy power.  If you are using batteries, be sure there is enough current capacity (max current draw).  A 9V battery is not sufficient.  And I hope you have connected the GPS VCC to a 5V supply, not a 3.3V supply.  Most modules have an on-board regulator that requires >3.6V at 70mA to operate correctly.

Some people have had trouble powering the GPS module from the Arduino 5V pin; it always works then both the Arduino and the GPS module to a separate 5V supply of sufficient capacity (> 0.5A).

Otherwise... it's the antenna.

aarush22, 2017-09-02:
It seems there was something wrong with the GPS module or the antenna, I got another module MT3333 gps module and it works quite nicely after a little adjustment. Thanks everyone for your help.

closed 2017-09-02 by aarush22