SlashDevin/NeoGPS #46: [Question] Sending UBX protocol commands

Redferne, 2017-09-04:
Trying to send a cold start request to a UBlox 7m device using the UBX protocol.
I figured this should work:

```
static const uint8_t gps_cold[] __PROGMEM = { ublox::UBX_CFG, ublox::UBX_CFG_RST, sizeof(ublox::cfg_reset_t), 0, 0xff, 0xff,    0x02,          0x00};
static const ublox::cfg_reset_t *cfg_cold_P = (const ublox::cfg_reset_t *) &gps_cold[0];

void gpsColdStart(void) {
  gps.send_request_P(*cfg_cold_P);
```
However, I do not get any response/reply on this command? :cry:

SlashDevin, 2017-09-05:
TBH, I've not tried putting the UBX message into PROGMEM.  I think there is an initialization problem, something to do with the constructor needing to "execute" instead of the storage being evaluated by the compiler.  You should be able to use `send_request`, though.  I'm working up an example...

SlashDevin, 2017-09-06:
Ok, there are three ways to send the reset command:

1. Construct a `cfg_reset_t` variable in RAM, set its members and use the `send_request` method.
2. Define a byte array in PROGMEM and send it with the utility function `sendUBX` (see ubloxRate.ino).
3. Cast the address of the PROGMEM byte array to a `cfg_reset_t *` and use `send_request_P`

Here is the code to do it all three ways:

```
static const uint8_t ubxReset[] __PROGMEM =
  { ublox::UBX_CFG, ublox::UBX_CFG_RST,
    UBX_MSG_LEN(ublox::cfg_reset_t), 0, // a word length... MSB is 0
    0x00, 0x00,    // Hotstart or...
    // 0x01, 0x00, // ... Warmstart or...
    // 0xff, 0xff, // ... Coldstart
    0x02,        // reset mode
    0x00         // reserved
  };

void resetGPS()
{
  const uint8_t WAY = 3; // pick 1,2 or 3
  
  switch (WAY) {
    case 1:
      {
        ublox::cfg_reset_t cfg_cold; // RAM local variable
        cfg_cold.clear_bbr_section =
          // Hotstart
          { false };

          // Warmstart
          // { true, false };

          // Coldstart
          // { true, true, true, true, true, true, true, true, true,
          //   true, // reserved1 is 2 bits
          //   true, true, true, 
          //   true, // reserved2 is 1 bit
          //   true };

        cfg_cold.reset_mode        = ublox::cfg_reset_t::CONTROLLED_SW_RESET_GPS_ONLY;
        cfg_cold.reserved          = 0;
        gps.send_request( cfg_cold );
      }
      break;

    case 2:
      sendUBX( ubxReset, sizeof(ubxReset) ); // just send the PROGMEM bytes
      break;

    case 3:
      {
        const ublox::cfg_reset_t *cfg_cold_ptr = (const ublox::cfg_reset_t *) ubxReset;
        gps.send_request_P( *cfg_cold_ptr );
      }
      break;
  }

} // resetGPS
```
There were two problems with the code you were trying:

1.  The UBX message length does not include the header bytes, so the byte array had the wrong length value (see helper macro [`UBX_MSG_LEN`](https://github.com/SlashDevin/NeoGPS/blob/master/src/ublox/ubxmsg.h#L78)).
2.  The is a bug in `send_request_P`.  It was not accessing the PROGMEM msgClass, msgID and length members correctly.  I've checked in a new version of ubxGPS.cpp with a fix.  This is not in the release yet, just get that one file.

closed 2017-09-06 by SlashDevin

Redferne, 2017-09-06:
I tested option 3 and it worked perfectly! Thank you! Now I will implement a power off command to save battery usage on my device. Cheers! :+1:

commit 2e3ace2b (2017-09-06): Fix #46 ublox `send_request_P`

*  Was not accessing PROGMEM members correctly