SlashDevin/NeoGPS #49: How to add data members to fix?

Rage441, 2017-10-10:
Hello,
Very impressed with the library, but it doesn't quite cover everything I need.  Using a Ublox GPS and I need most of the data from the PUBX_00 message, but have been unable to get the vertical velocity.  Have tried going through the various .h and .cpp files to add a "GPS_FIX_VERT_VEL" and "parseVVel()", but have been unsuccessful.  Can this be easily done?  Is there a set of steps I should follow to incorporate this?  Brand new to GitHub so my apologies if this isn't the right forum/format.

SlashDevin, 2017-10-10:
To add a new fix member:

*  Add the option to GPSfix_cfg.h
*  Add the data member and valid flag to GPSfix.h
*  Declare a parse routine to ubxNMEA.h
*  Call the parse routine from a new case in the switch statement in `parsePUBX_00` (ubxNMEA.cpp).
*  Implement the parse routine in ubxNMEA.cpp.  Be sure to model it on other parse routines that produce the same data type.  Wrap it with your new option, call the appropriate data type parser and set the validity flag when completed.  There are some new methods for performing character and field validation.

Post what you've got when you're done (or if you have questions), and I'll put it on the list for enhancements.

Rage441, 2017-10-10:
Wow,

Thanks for the amazingly quick reply.  Will let you know how it goes.

Reagh

Rage441, 2017-10-10:
Apologies for the lengthy code snippets below.  I get the following error when trying to compile PUBX.ino (which worked before making changes to the library files for the vertical velocity).  I suspect it’s something relatively simple that I’ve missed, but I’m stumped at this point.

C:\...\NeoGPS\src\NMEAGPS.cpp:1702:38: error: no 'bool NMEAGPS::parseVertVel(char)' member function declared in class 'NMEAGPS'

bool NMEAGPS::parseVertVel( char chr )

                                      ^

Error compiling for board Teensy 3.6.
                                      ^
I’ve made the following changes (in blue) to the noted files:

PS – I use the “###” in the comments to denote developmental changes so I can easily find them later.

============================================================
Changes to GPSfix_cfg.h

#define GPS_FIX_SPEED
#define GPS_FIX_VERT_VEL  // ### New
#define GPS_FIX_HEADING

============================================================
Changes to GPSfix.h

"Members of GPS Fix";
Added after GPS_FIX_SPEED:

  #ifdef GPS_FIX_VERT_VEL    //### NEW
    whole_frac    vspd; // .001 metres per second

    uint32_t vvel_mkn () const { return vspd.int32_000(); };
    float    vvel     () const { return vspd.float_000(); };

    // Utilities for vertical velocity in other units
    //CONST_CLASS_DATA float FT_PER_M = 3.2808;
    //float    vvel_fts () const { return vvel() * FT_PER_M; };
    //float    vvel_ftmin () const { return vvel_fts () * 60; };

  #endif

****

  //  Flags to indicate which members of this fix are valid.
Added after GPS_FIX_SPEED

    /// ### NEW
    #ifdef GPS_FIX_VERT_VEL
      bool vvel NEOGPS_BF(1);
    #endif

****

//  Initialize a fix.
Added after GPS_FIX_SPEED

    /// ### NEW
    #ifdef GPS_FIX_VERT_VEL
      vspd.init();
    #endif

****

// Merge valid fields.
Added after GPS_FIX_SPEED

    // ### NEW
    #ifdef GPS_FIX_VERT_VEL
      if (r.valid.vvel)
        vspd = r.vspd;
    #endif

============================================================
Changes to ubxNMEA.h

// NMEA 0183 Parser for ublox Neo-6 GPS Modules.
Added near end; after    "bool parseFix( char chr );"

    bool parseVertVel( char chr );  // ### New

============================================================
Changes to ubxNMEA.cpp
case 13 added

    #ifdef NMEAGPS_PARSE_PUBX_00
      case  2: return parseTime( chr );
      PARSE_LOC(3);
      case  7: return parseAlt( chr );
      case  8: return parseFix( chr );
      case 11: return parseSpeed( chr ); // kph!
      case 12: return parseHeading( chr );
      case 13: return parseVertVel( chr );  // ### New
      case 15: return parseHDOP( chr );
      case 16: return parseVDOP( chr );
      case 18: return parseSatellites( chr );
    #endif

============================================================
Changes to NMEAGPS.cpp
essentially copied the code for speed and used it for Vertical Velocity
Added after "parseSpeed"

// ### NEW

bool NMEAGPS::parseVertVel( char chr )
{
  #ifdef GPS_FIX_VERT_VEL
    if (chrCount == 0)
      NMEAGPS_INVALIDATE( vvel );
   if (parseFloat( m_fix.vspd, chr, 3 )) {

      if (validateFields() && m_fix.valid.vvel && negative)
        sentenceInvalid();
      else
        m_fix.valid.vvel = (chrCount != 0);
    }
  #endif

  return true;

} // parseVertVel

NOTE:  I tried adding the above “parseVertVel( char chr )” to ubxNMEA.cpp but it kicked a large number of errors so I essentially duplicated what was used for “parseSpeed” in NMEAGPS.cpp

============================================================
Changes to Streamers.cpp
added after GPS_FIX_SPEED

  #if defined(GPS_FIX_VERT_VEL)
    "vvel,"
  #endif

****

    //  Date/Time not enabled, just output the interval number

    #ifdef GPS_FIX_VERT_VEL
      if (fix.valid.vvel)
        outs.print( fix.vvel(), 3 ); // m/s
      outs << ',';
    #endif

****

// not USE_FLOAT ----------------------

    #ifdef GPS_FIX_VERT_VEL
      if (fix.valid.vvel)
        outs << fix.vvel_mkn();
      outs << ',';
    #endif

============================================================

SlashDevin, 2017-10-11:
```
bool NMEAGPS::parseVertVel( char chr )
```

That is a member of ubloxNMEA, not NMEAGPS.  The the declaration should be in ubxNMEA.h (as per your description above), and its implementation should be in ubxNMEA.cpp:

```
bool ubloxNMEA::parseVertVel( char chr )
        ^
```

Everything else looks good!

closed 2017-10-18 by SlashDevin