SlashDevin/NeoGPS #50: NeoGPS Compile "pgm_read_ptr" redefined on T3.5

mjs513, 2017-10-11:
Using IDE 1.8.5 and TD1.40 I tried compiling a program I have using NeoGPS and I received the following warning message:
```
Programs\arduino-1.8.5\hardware\teensy\avr\cores\teensy3/WString.h:33:0,
                 from C:\Local Programs\arduino-1.8.5\hardware\teensy\avr\cores\teensy3/Print.h:38,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\DMS.cpp:20:
C:\Local Programs\arduino-1.8.5\hardware\teensy\avr\cores\teensy3/avr/pgmspace.h:113:0: warning: "pgm_read_ptr" redefined
 #define pgm_read_ptr(addr) ({ \
 ^

In file included from C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\DMS.h:21:0,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\DMS.cpp:18:
C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\NeoGPS_cfg.h:106:0: note: this is the location of the previous definition
   #define pgm_read_ptr(addr) (*(const void **)(addr))
 ^

```
I reported this as an issue in Teensy core-libs and Paul identified the problem as in NEOGPS as associated with:

```
//------------------------------------------------------------------------
// The PROGMEM definitions are not correct for Zero and MKR1000

#if !defined(__AVR__)
  // TODO: use the Zero/MKR1000-specific symbols
  #undef pgm_read_ptr
  #define pgm_read_ptr(addr) (*(const void **)(addr))
#endif
```

Guess easy fix would be:
```
#if !defined(__AVR__)  to #if !defined(__AVR__) or !defined(__TEENSYDUINO__)
```
Thanks
Mike

SlashDevin, 2017-10-11:
...and the TO DO becomes a MUST DO...

Those other platforms are a little behind, so I'll have to look up the build-specific symbols for them.  Sigh.

mjs513, 2017-10-11:
It currently just gives a warning.  It does compile and works with the redefinition, so not an urgency on my part :)

commit 08201f1a (2017-10-12): Fix #50, pgm_read_ptr warning
closed 2017-10-12 by SlashDevin