SlashDevin/NeoGPS #52: Show adjust date & time in LCD 5110

beryindo, 2017-10-14:
how to show adjust date & time in LCD 5110 ?
I try not show in LCD 5110
`  while (gps.available( gpsPort )) {
    fix = gps.read();
    // Display the local time

    if (fix.valid.time && fix.valid.date) {
      adjustTime( fix.dateTime );

      display << fix.dateTime;
    }
    display.display();
  }`

SlashDevin, 2017-10-14:
If you don't have a valid date/time yet, nothing will display.  You could display dashes if there is no valid date/time:

```
if (fix.valid.time && fix.valid.date) {
  adjustTime( fix.dateTime );
  display << fix.dateTime;
} else {
  display.print( F("------") );
}
display.display();
```

This would confirm that you do not have a valid date/time.

Also, be sure to set the cursor position before printing the date/time.  There are different ways to do this in different libraries.   It's usually something like `setCursor` or `moveTo`.

You can also print the pieces in any order you want:

```
      if (fix.dateTime.hours < 10)
        display.print( '0' );
      display.print(fix.dateTime.hours);
      display.print( ':' );
      if (fix.dateTime.minutes < 10)
        display.print( '0' );
      display.print(fix.dateTime.minutes);
      display.print( ':' );
      if (fix.dateTime.seconds < 10)
        display.print( '0' );
      display.print(fix.dateTime.seconds);
```

This is an excerpt from the example NMEASDlog.ino.

beryindo, 2017-10-14:
how to set GMT +7 ?

SlashDevin, 2017-10-15:
Look at NMEAtimezone

beryindo, 2017-10-16:
```
#if !defined(GPS_FIX_TIME) | !defined(GPS_FIX_DATE)
  #error You must define GPS_FIX_TIME and DATE in GPSfix_cfg.h!
#endif

#if !defined(NMEAGPS_PARSE_RMC) & !defined(NMEAGPS_PARSE_ZDA)
  #error You must define NMEAGPS_PARSE_RMC or ZDA in NMEAGPS_cfg.h!
#endif

// Set these values to the offset of your timezone from GMT

static const int32_t          zone_hours   = +7L; // EST
static const int32_t          zone_minutes =  0L; // usually zero
static const NeoGPS::clock_t  zone_offset  =
                                zone_hours   * NeoGPS::SECONDS_PER_HOUR +
                                zone_minutes * NeoGPS::SECONDS_PER_MINUTE;

void adjustTime( NeoGPS::time_t & dt )
{
  NeoGPS::clock_t seconds = dt; // convert date/time structure to seconds

  //  First, offset from UTC to the local timezone
  seconds += zone_offset;

  #ifdef CALCULATE_DST
    //  Then add an hour if DST is in effect
    if ((springForward <= seconds) && (seconds < fallBack))
      seconds += NeoGPS::SECONDS_PER_HOUR;
  #endif

  dt = seconds; // convert seconds back to a date/time structure

} // adjustTime

void loop() {
  while (gps.available( gpsPort )) {
    fix = gps.read();
    // Display the local time

    if (fix.valid.time && fix.valid.date) {
      adjustTime( fix.dateTime );

      display << fix.dateTime;
    }
    display.display();
  }
}
```

can't show in 5110

SlashDevin, 2017-10-16:
Did you read this:

>Also, be sure to set the cursor position before printing the date/time. There are different ways to do this in different libraries. It's usually something like setCursor or moveTo.

If you don't read my responses, I won't waste time writing them.

closed 2017-10-18 by SlashDevin