SlashDevin/NeoGPS #54: 10hz neo m8n

beryindo, 2017-10-16:
how to set 10hz neo m8n ?

SlashDevin, 2017-10-16:
Look at ubloxRate.ino for code examples.  It shows how to send various commands.  I **do not** recommend that you build and upload this sketch.  Just **look** at the definition of the binary commands (i.e., `ubxRate10Hz`) and how to send them (i.e., `sendUBX`).

You should also get familiar with the [ublox NEO-M8 Protocol Spec](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_(UBX-13003221)_Public.pdf).  It describes **all** the commands you can send, either with NMEA Proprietary messages ($PUBX, section 30.3) or UBX binary messages (CFG-xxx, section 31.11).  You can see some of those commands in ubloxRate.ino

beryindo, 2017-10-16:
I am use sim900 for send data tcp. https://github.com/Seeed-Studio/GPRS_SIM900
but sim900 can't 
```
gprs.print(fix.latitudeDMS.NS())
fix.latitudeDMS.printDDDMMmmmm(gprs);
```

how to fix it ?

beryindo, 2017-10-18:
ubloxRate.ino:2:26: fatal error: NeoTeeStream.h: No such file or directory

beryindo, 2017-10-18:
thank you, solved

closed 2017-10-18 by SlashDevin

23100048, 2024-03-13:
> ubloxRate.ino:2:26: fatal error: NeoTeeStream.h: No such file or directory where I can get NeoTeeStream.h ?

I am getting this same error. Can you kindly share the solution?

23100048, 2024-03-13:
> > ubloxRate.ino:2:26: fatal error: NeoTeeStream.h: No such file or directory where I can get NeoTeeStream.h ?
> 
> I am getting this same error. Can you kindly share the solution?

The solution to this problem is to download both header files (Neotee.h and Neoteestream.h) from https://github.com/SlashDevin/NeoTee/tree/master into the src folder. Apparently this is an issue (amongst others) on the arduino mega boards.