SlashDevin/NeoGPS #55: GPS data in Minimum Time

JamesPwi, 2017-10-17:
Hey SlashDevin

i have not tested this code but i have one query and hoping you will help me with that 

Query is : i need Long , Lat and HDOP data cont. from GPS within minium time like 1uh (microsecond) so is there  a way that we can use your library and  we get all those data is less time ?

SlashDevin, 2017-10-18:
>I need Long , Lat and HDOP data cont. from GPS within minimum time like 1us

I doubt that.  What are you trying to do?

JamesPwi, 2017-10-18:
i am working on  digital unmanned vehicle and working on one of the feature called "Return to Home" . i already prepared one code but its taking 1millisecond time  and that time making my vehicle unbalanced so i am looking for some solution where i get that data in very less time so that my vehicle become stable when i access all my codes together

SlashDevin, 2017-10-18:
1ms would be fine for IMU and sensor sampling, but not for GPS updates.  The most common (cheapest) GPS devices can provide updates at 100ms.  Expensive GPS devices can provide updates at 10ms.

NeoGPS will parse the GPS data much faster than it arrives, using less than 1ms of CPU time per sentence.  However, it takes time to _receive_ an update from the GPS device.  At 9600 baud, your Arduino receives one GPS character in 1ms.  It takes about 72 characters to describe the GPS update.

If you research autonomous vehicle software, you will see that the GPS and compass update rates are always much slower than IMU or wheel sensor update rates.  The sketch performs "math" to integrate these different sensor feeds.

JamesPwi, 2017-10-19:
yes all your information is correct  , still i am looking how i can reduce my time as less as i can. so can you please  help me out  as i only need 3 things, longitude , latitude and HDOP value .

JamesPwi, 2017-10-20:
my gps refress rate is 10 Hz and when i am getting lat and lon through library in arduino 2560 i am getting 900 micro seconds of loop time.Is there is any way to get less loop time than this to get same data through same micro controller or use of interrupt?

SlashDevin, 2017-10-20:
>  i only need 3 things, longitude , latitude and HDOP value

Then there should only be 2 **un**commented fields in GPSfix_cfg.h:

```
#define GPS_FIX_LOCATION
#define GPS_FIX_HDOP
```

Everything else should be commented out.

And I would question whether you really need HDOP.  Remember that you should always check `fix.valid.location` before using `fix.latitude()` or `fix.longitude()`.  You can use HDOP as a basic rejection test, but if you need to know the actual error distance (estimate in cm), you'll need to enable those fields instead and use the GST sentence.

> when i am getting lat and lon through library in arduino 2560 i am getting 900 micro seconds of loop time.

Are you using a NeoGPS example, or some other library's example?  Other libraries' examples are usually not structured properly.  Please post the smallest example that shows this loop time.  Be sure to put \``` before and after the code so it posts like my snippet above.

JamesPwi, 2017-10-21:
I run the gps simple example of neo gps library in arduino 2560 and i getting the 900 micro seconds loop time but i want the loop time of around 400 micro seconds.Is this is possible through neo gps library? 
Secondly i want hdop because i want to save the starting valid location in a variable in void setup of arduino code.Once i get valid location then void loop of my code will run.So in void setup i will write the code as 
"while(hdop<= 2.0)
{
Save latitude and longitude in a variable
}"

SlashDevin, 2017-10-22:
> the gps simple example [has] 900us loop time

I'll have to see how you are timing it, but I suspect you are timing the printing operations, especially the floating-point print of lat/lon.

As I said before, the **total** time used by NeoGPS to parse one sentence is less than 1ms.  That 1ms CPU time is spread out over the sentence receive time, about 72ms (for a 72-character GPRMC sentence at 9600) during the 1s GPS update interval (for 1Hz update rate).

> Once i get valid location then `loop` will run.<br>
> So in `setup` i will write the code as<br>
> ```
> while(hdop<= 2.0)
> {
>   Save latitude and longitude in a variable
> }
> ```

I think you want to do something like this:

```
NeoGPS::Location_t startLocation;

void setup()
{
  gpsPort.begin( ... );

  for (;;) {
    if (gps.available( gpsPort )) {
      gps_fix firstFix = gps.read();
      if (firstFix.valid.location && firstFix.valid.hdop && (firstFix.hdop < 2000)) {
        startLocation = firstFix.location;
        break;
      }
    }
  }
```
The HDOP is scaled by 1000 so it can be stored in an integer.  The Data Model page describes each value.  You can also look at the header file, GPSfix.h.

No values in the `fix` structure are stored as floating-point numbers, not even lat/lon.  When you call the `fix.latitude()` function, it converts the internal integer form to a floating-point number.  This is to avoid the slow floating-point math, to reduce the program size (no floating-point library routines linked in), and preserve the accuracy of the GPS information (other GPS libraries use floating-point numbers, which is limited to 6 significant digits).

JamesPwi, 2017-10-28:
Hey SlashDevin

last time you  show me one sample code of HDOP

NeoGPS::Location_t startLocation;

void setup()
{
  gpsPort.begin( ... );

  for (;;) {
    if (gps.available( gpsPort )) {
      gps_fix firstFix = gps.read();
      if (firstFix.valid.location && firstFix.valid.hdop && (firstFix.hdop < 2000)) {
        startLocation = firstFix.location;
        break;
      }
    }
  }

 i have one query on this , do we need to make few changes in your library so that "firstfix.hdop" will started when we shart our code  or this will work directly with your library.

And Second thing is  : like i  have two " latitude and longitude position" and i want that distance to be calculated in feet. so is there a way in your library which can help us on this  ?

SlashDevin, 2017-10-30:
> do we need to make few changes in your library so that "firstfix.hdop" will start when we start our code

No, the code must wait for the GPS device to send a sentence with HDOP in it.  You can't **force** the GPS to send anything, you have to wait.  Even if the GPS device started sending HDOP when the sketch started, your sketch has to wait about 80ms to receive all the characters over the serial connection.

You could put a timeout in `setup`, if you want to continue.  Or, you could just let loop check HDOP every time a fix is received:

```
void setup()
{
  gpsPort.begin( 9600 );
}

void loop()
{
  if (gps.available( gpsPort )) {
    fix = gps.read();

    if (fix.valid.hdop && (fix.hdop < 2000)) {
      // HDOP is good enough

      digitalWrite( LED_PIN, HIGH ); // turn LED on to show that we have a good fix

      // use the fix to do something, once per second

    } else {
      digitalWrite( LED_PIN, LOW ); // turn LED off to show that we do not have a good fix

      // HDOP too big, don't use it.
    }
  }

  ... other things you need to check here...
}
```

JamesPwi, 2017-10-30:
can you please let me know on my other question too : 

if I have two " latitude and longitude position" and i want that distance to be calculated in feet. so Is there a way in your library which can help us on this?

SlashDevin, 2017-10-30:
Try searching for "distance" in the search box at the top of the page.  It will show matches in the [Location document page](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Location.md#distance) and in Location.H and CPP.  Just convert from km to feet.

closed 2017-11-10 by SlashDevin