SlashDevin/NeoGPS #57: 10ghz without input r0

beryindo, 2017-10-27:
how to set gps to 10ghz without input r0 sir ?

SlashDevin, 2017-10-30:
> how to set gps to 10ghz

You can't.  The GPS satellite radio transmission frequency is only 1.5GHz, so 10GHz is ridiculous.

Furthermore, most Arduinos run at 8MHz, 16MHz or 20MHz.  Some run at 80MHz or a little higher.  No Arduinos run at 1GHz.  You do not have an Arduino that could process 1GHz GPS updates.

GPS update rates are usually no faster than 10Hz.  You asked about this rate in #54.  If you spend more money, you can get 20Hz or even 100Hz.  Most applications use interpolation to "calculate" extra positions between 1Hz GPS updates.

> how to set baud rate 115200 without input r0?

Use an Arduino that has extra hardware serial ports, like a Mega, Micro, Due or Teensy.  Arduino sketches can use those additional ports through variables like `Serial1`, `Serial2` or `Serial3`.  Software serial ports, using `SoftwareSerial`, `NeoSWSerial` or `AltSoftSerial`, are not reliable at that baud rate.  `AltSoftSerial` might work, if the crystal frequency is accurate, and your sketch does not do anything else but read GPS data.

beryindo, 2017-11-03:
oh sorry sir, I mean 10hz :+1:

closed 2017-11-04 by SlashDevin