SlashDevin/NeoGPS #58: ESP32 support

nsollars, 2017-11-10:
Any chance the esp32 will get support soon ( noticed you have esp8266 there ) ...

Thanks

SlashDevin, 2017-11-10:
I did not know that the esp32 is a different target from the esp16 series (e.g., and esp8266).  Have you tried building any of the examples?

nsollars, 2017-11-10:
Hi,

Yes I have seems to be some difference with progmem,  I tested out the sam-m8q I got with a Mega just to get it going but compiling with esp32 ( say nmeaSimple )

WARNING: library NeoGPS claims to run on (avr, samd, sam, esp8266, arc32) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
WARNING: library AltSoftSerial claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
/home/nsollars/Arduino/libraries/NeoGPS/examples/NMEAsimple/NMEAsimple.ino: In function 'void loop()':
NMEAsimple:67: error: 'struct gps_fix::valid_t' has no member named 'altitude'
     if (fix.valid.altitude)
                   ^
NMEAsimple:68: error: 'class gps_fix' has no member named 'altitude'
       DEBUG_PORT.print( fix.altitude() );
                             ^
exit status 1
'struct gps_fix::valid_t' has no member named 'altitude'

And for the UBlox example

WARNING: library NeoGPS claims to run on (avr, samd, sam, esp8266, arc32) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
WARNING: library AltSoftSerial claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
In file included from /home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.h:26:0,
                 from /home/nsollars/Arduino/libraries/NeoGPS/examples/ublox/ublox.ino:2:
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:179:34: warning: width of 'ublox::cfg_rate_t::time_ref' exceeds its type
         enum time_ref_t time_ref:16;
                                  ^
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:176:12: note: offset of packed bit-field 'ublox::cfg_rate_t::time_ref' has changed in GCC 4.4
     struct cfg_rate_t : msg_t {
            ^
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp: In member function 'NMEAGPS::decode_t NMEAGPS::parseCommand(const NMEAGPS::msg_table_t*, uint8_t, char)':
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp:564:89: error: 'pgm_read_ptr' was not declared in this scope
         const char * const *table   = (const char * const *) pgm_read_ptr( &msgs->table );
                                                                                         ^
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp:605:68: error: 'pgm_read_ptr' was not declared in this scope
         msgs = (const msg_table_t *) pgm_read_ptr( &msgs->previous );
                                                                    ^
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp: In member function 'const __FlashStringHelper* NMEAGPS::string_for(NMEAGPS::nmea_msg_t) const':
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp:646:89: error: 'pgm_read_ptr' was not declared in this scope
         const char * const *table   = (const char * const *) pgm_read_ptr( &msgs->table );
                                                                                         ^
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp:655:66: error: 'pgm_read_ptr' was not declared in this scope
       msgs = (const msg_table_t *) pgm_read_ptr( &msgs->previous );
                                                                  ^
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp: In static member function 'static void NMEAGPS::poll(Stream*, NMEAGPS::nmea_msg_t)':
/home/nsollars/Arduino/libraries/NeoGPS/src/NMEAGPS.cpp:1986:82: error: 'pgm_read_ptr' was not declared in this scope
       (const __FlashStringHelper *) pgm_read_ptr( &poll_msgs[msg-NMEA_FIRST_MSG] );
                                                                                  ^
exit status 1
Error compiling for board WEMOS LOLIN32.

Ive gotten the device going on the esp32 using the hardwareserial <Serialx>(2) method for this one so it seems to be a case of cleaning this up ..  The library is under the unsupported section as per the properties also.

Hope this helps

SlashDevin, 2017-11-11:
Ok, I don't think it's really incompatible.  Here are the two key issues:

1) The `library.properties` needs "esp32" added to the `Architectures=` line.  That would eliminate most of the architecture warnings.  I will add this if you're willing to try it.

2) The ESP32 core does not have the `pgm_read_ptr` macro defined *yet*.  This deficiency has been corrected in all the other Arduino cores (e.g. SAM and SAMD).  You could move this line in NeoGPS_cfg.h outside of the conditional `#if`:

```
//------------------------------------------------------------------------
// The PROGMEM definitions are not correct for Zero, MKR1000 and
//    earlier versions of Teensy boards

//#if defined(ARDUINO_SAMD_MKRZERO) | defined(ARDUINO_SAMD_ZERO) |   \  <-- comment out
//    (defined(TEENSYDUINO) && (TEENSYDUINO < 139))                     <-- comment out
  #undef pgm_read_ptr
  #define pgm_read_ptr(addr) (*(const void **)(addr))
//#endif                                                                <-- comment out
```

This will define the macro for your build.  If this works, it should be reported as a bug to the ESP32 maintainers.

For this error:

```
 error: 'struct gps_fix::valid_t' has no member named 'altitude'
```

You must have commented out the altitude item in `GPSfix_cfg.h`.  Make sure this line is not commented:

```
#define GPS_FIX_ALTITUDE
```

I think that should allow all the examples to build properly.  Let me know.

nsollars, 2017-11-12:
Hello,

Yes that seems have fixed up the issue,  so I also removed some earlier things I did with regards to software serial ( altserial ) and also commented out that one from line 160 in GPSport,h.

So here is the output of the build,

In file included from /home/nsollars/Arduino/libraries/NeoGPS/examples/ublox/ublox.ino:43:0:
/home/nsollars/Arduino/libraries/NeoGPS/src/GPSport.h:199:6: warning: #warning Using Serial for GPS connection. [-Wcpp]
     #warning Using Serial for GPS connection.
      ^
In file included from /home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.h:26:0,
                 from /home/nsollars/Arduino/libraries/NeoGPS/examples/ublox/ublox.ino:2:
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:179:34: warning: width of 'ublox::cfg_rate_t::time_ref' exceeds its type
         enum time_ref_t time_ref:16;
                                  ^
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:176:12: note: offset of packed bit-field 'ublox::cfg_rate_t::time_ref' has changed in GCC 4.4
     struct cfg_rate_t : msg_t {
            ^
Sketch uses 149278 bytes (11%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11460 bytes (3%) of dynamic memory, leaving 283452 bytes for local variables. Maximum is 294912 bytes.

Nige

nsollars, 2017-11-12:
ok so the only issue now is to wire in if defined esp32 we have to create 

HardwareSerial Serial1(2) <-- the 3rd hardware serial we can use ( pins 16 and 17 ) ..

Serial0 and 1 on the esp32 are tied to the usb and flash ( internal ).

Nige

SlashDevin, 2017-11-12:
If I understand what you mean by "create HardwareSerial2", you can delete the entire contents of GPSport.h and replace it with just a few lines that force all the examples to use the serial port you choose (described [here](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Installing.md#choosing-your-own-serial-port)).

The default GPSport.h tries to guess what serial port should be used.  It does not work in your case.

BTW, be sure to start with the simpler examples before trying the ublox binary protocol examples.  There are very few reasons that you would really need the UBX protocol.

nsollars, 2017-11-12:
Alright, sounds good il give a few things a try and then get back to you..

nsollars, 2017-11-12:
Alright,

So after mauling it over here is what I did after removing everything down to the debug port defines,

HardwareSerial esp32HWSerial(2);

#define GPS_PORT_NAME esp32HWSerial
#define gpsPort esp32HWSerial

NMEASimple now works as expected, just as some info on compilation there are a number of warnings.  I am guessing that youl want to clean those up,

-- Ouput

In file included from /home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.h:26:0,
                 from /home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.cpp:22:
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:179:34: warning: width of 'ublox::cfg_rate_t::time_ref' exceeds its type
         enum time_ref_t time_ref:16;
                                  ^
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:176:12: note: offset of packed bit-field 'ublox::cfg_rate_t::time_ref' has changed in GCC 4.4
     struct cfg_rate_t : msg_t {
            ^
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.cpp: In member function 'virtual bool ubloxGPS::parseField(char)':
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.cpp:460:65: warning: comparison between 'enum NMEAGPS::nmea_msg_t' and 'enum ubloxGPS::ubx_msg_t' [-Wenum-compare]
   if ((nmeaMessage != NMEAGPS::NMEA_UNKNOWN) && (nmeaMessage != UBX_MSG))
                                                                 ^
In file included from /home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxGPS.h:26:0,
                 from /home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.cpp:18:
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:179:34: warning: width of 'ublox::cfg_rate_t::time_ref' exceeds its type
         enum time_ref_t time_ref:16;
                                  ^
/home/nsollars/Arduino/libraries/NeoGPS/src/ublox/ubxmsg.h:176:12: note: offset of packed bit-field 'ublox::cfg_rate_t::time_ref' has changed in GCC 4.4
     struct cfg_rate_t : msg_t {
            ^
Sketch uses 144230 bytes (11%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11476 bytes (3%) of dynamic memory, leaving 283436 bytes for local variables. Maximum is 294912 bytes.

--\ output

Thanks for you help man, really appreciate it.  If you need any more info / testing from me dont hesitate to reach out.

Nige

SlashDevin, 2017-11-12:
The `time_ref` warnings are ok... they appeared with the newest version of gcc, and I suspect they will go away with the next release.

The warning about comparing enums is something I'll look at.  Some casting is probably required.

closed 2017-11-12 by SlashDevin