SlashDevin/NeoGPS #59: Fix Status vs Fix Quality

mjs513, 2017-12-04:
Hi @SlashDevin 

I am making an assumption that fix status would be the equivalent of Fix Quality.   It looks like fix.status supports 0, 1, 2 and 6.  Would like to know if it is possible to add the following enumerations to the library:
                               3 = PPS fix
			       4 = Real Time Kinematic
			       5 = Float RTK

Thanks
Mike

SlashDevin, 2017-12-05:
I just had a request [over here](https://forum.arduino.cc/index.php?topic=305867.msg3511282#msg3511282) for values 4 & 5, so I was wondering what value 3 was for...  Thanks, I'll take a look.

mjs513, 2017-12-05:
According to this http://resources.esri.com/help/9.3/ArcGISmobile/adf/ESRI.ArcGIS.Mobile~ESRI.ArcGIS.Mobile.Gps.GpsFixStatus.html, PPSFix is defined as:

> PPSFix indicates that Precise Positioning System, encrypted for government use is being used by the receiver. PPS is the most accurate positioning, velocity, and timing information continuously available, worldwide, from the basic GPS. This service will be limited to authorized U.S. and allied Federal Governments; authorized foreign and military users; and eligible civil users. Unauthorized users will be denied access to PPS through the use of cryptography.

I am assuming the Ublox M8P would give a status of 3 for a valid fix. Figured if you are going to add 4 and 5  might as well as add 3 to avoid making another mod.

By the way finally got around to hook up everything up using RTKlib transmitting actual data over a xbee to a Teensy and your library handles it very well.  Just need to longitudeL to get the accuracy.

Mike

mjs513, 2017-12-11:
Wow that was fast.  Will download and give it a try and give it a try soon.   I will close this and if there are any issues will let you know. 

Thanks
Mike

closed 2017-12-11 by mjs513

guyostfeld, 2018-01-29:
Can you update documentation and ENUM ? we are not 100% sure we understand the outcome of this discussion.

SlashDevin, 2018-02-04:
Hmm, I'm not sure what else you need.  There is one place on the Data Model page that mentions status.  It should be updated, but there is no other documentation outside of the GPSfix.h file.

As mentioned in GPSfix.h, the status values are sorted from "least accurate" to "best", so you can compare `fix.status` against the minimum you want to accept.  This is completely independent from the `fix.valid` flags and the `DOPs`.

Most applications do not care about the status; they usually just need to check `fix.valid.location` before using the lat/lon values.  Next most common is comparing the DOPs against some minimum quality value (e.g., 2.000, which is stored as 2000 in `fix.hdop`).  You could also use the DOP to provide an "uncertainty" indicator.  DOPs are independent from the GPS status.

To show the estimated error in a location, you have to use the GST sentence to obtain the lat/lon/alt values.  This is also independent from the current GPS status.

In the context of this Issue, GPS devices with advanced error-reduction techniques can provide insight into their progress in reducing the typical 3m-10m errors.  The additional status values indicate that errors have been reduced by increasing amounts.