SlashDevin/NeoGPS #61: NeoGPS with RTK Solution Output as NMEA message

mjs513, 2017-12-04:
Hi SlashDevin

This is only an issue if you are using the output of rtknavi in nmea format.  The standard GGA message for lat  is ddmm.mmmm from the ublox module.  In the rtknavi message its ddmm.mmmmmmm (eg., 4046.4769300) or longitude dddmm.mmmmmmm eg., 07348.8767718.  Is there a simple way to get the added precsion?

Thanks
Mike

mjs513, 2017-12-05:
My mistake.  Dumped the transmitted messages to a file as well as transmitting them and then did the comparison.  When I did it that way all was well.  My apologies.

closed 2017-12-05 by mjs513

SlashDevin, 2017-12-05:
I'd like to comment on this good question:

> In the rtknavi message its ddmm.mmmmmmm

That is 11 significant digits, which exceeds the 10 available in a 32-bit integer.  I have occasionally considered extra storage for those extra digits, but it's only necessary when you have a high-precision solution.  The UBX binary messages have extra fields for this reason.

Until then, NeoGPS *should* ignore those extra digits when parsing.  You would lose them if you store the truncated integer instead of the original "ddmm.mmmmmmm" characters,