SlashDevin/NeoGPS #62: Raw NMEA and fixed Data - no date/time

thomaszietz, 2017-12-07:
Hi,

I have to use the raw NMEA-String like "$GPGSA,A,3,04,05,,09,12,,,24,,,,,2.5,1.3,2.1*39" and the fixed datas (date, time, Lat......).

It works fine, but if I use "Serial.print(c) in GPSLoop (see below), I never get a valid date/time.

Where do I make a mistake?

Thomas

static void GPSloop()
{

  while (gpsPort.available()) {
    char c = gpsPort.read();
    Serial.print(c);                              //<<< No valid date/time if this is not commented out

    if (gps.decode( c ) == NMEAGPS::DECODE_COMPLETED) {

      Serial.println("");

      if (gps.fix().valid.time && gps.fix().valid.date) {

        adjustTime( gps.fix().dateTime );
        DEBUG_PORT.print("Date/Time: " );
        DEBUG_PORT << gps.fix().dateTime;
      }
      else{
        DEBUG_PORT.print("no Date/Time" );
      }

      DEBUG_PORT.print(" - Speed km/h: " );
      DEBUG_PORT.print( gps.fix().speed_kph(), 0 );
      DEBUG_PORT.print(" - Höhe m: " );
      DEBUG_PORT.print( gps.fix().altitude(), 0 );
      DEBUG_PORT.print(" - fix.satellites: " );
      DEBUG_PORT.print( gps.fix().satellites );
      DEBUG_PORT.print(" - LAT: ");
      DEBUG_PORT.print( gps.fix().latitude(), 4 ); // floating-point display
      DEBUG_PORT.print(" - LON: ");
      DEBUG_PORT.print( gps.fix().longitude(), 4 ); // floating-point display
      //DEBUG_PORT.println();

    }

  }

} // GPSloop

SlashDevin, 2017-12-07:
There are two problems:

1) Your device probably sends more than one kind of sentence (not just GSA).  Not all sentences have date and time.  You should not use the character-oriented method in this case.  The fix-oriented methods will merge all sentences into one fix, so you would get a date & time in every fix, once per update (usually 1Hz).  Your current loop will process each NMEA sentence, usually 5 sentences per update.

2) You are [printing too much data](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Troubleshooting.md#printing-too-much).  Echoing the raw NMEA characters almost always causes this problem.  And printing over 100 characters does not help.  This is my **guess**, because I can't see the `gpsPort` declaration, nor the baud rates.

It is also a little dangerous to use `gps.fix()` after calling `Serial.print`.  The print may block, allowing time for more GPS characters to arrive.  This could put the special `gps.fix()` in an undefined or initial state.

I would strongly suggest this instead:

```
static void GPSloop()
{

  while (gps.available( gpsPort )) {
    gps_fix fix = gps.read();

    if (fix.valid.time && fix.valid.date) {
      adjustTime( fix.dateTime );
      DEBUG_PORT.print( F("Date/Time: ") );
      DEBUG_PORT << fix.dateTime;
    }
    else{
      DEBUG_PORT.print( F("no Date/Time") );
    }

    DEBUG_PORT.print( F(" - Speed km/h: ") );
    DEBUG_PORT.print( fix.speed_kph(), 0 );
    DEBUG_PORT.print( F(" - Höhe m: ") );
    DEBUG_PORT.print( fix.altitude(), 0 );
    DEBUG_PORT.print( F(" - fix.satellites: ") );
    DEBUG_PORT.print( fix.satellites );
    DEBUG_PORT.print( F(" - LAT: ") );
    DEBUG_PORT.print( fix.latitude(), 4 ); // floating-point display
    DEBUG_PORT.print( F(" - LON: ") );
    DEBUG_PORT.print( fix.longitude(), 4 ); // floating-point display
    //DEBUG_PORT.println();

  }

} // GPSloop
```

SlashDevin, 2017-12-08:
Well, you might be able to do this:

```
void loop()
{
  if (gpsPort.available()) {
    char c = gpsPort.read();

    bt.write( c ); // forward raw NMEA data to Android tablet

    gps.handle( c ); // parse one character (ignore return value)

    if (gps.available()) {
      gps_fix fix = gps.read();

      // display fix data on the LCD
         ...
    }
  }
}
```
The bluetooth baud rate must be >= the GPS baud rate.

SlashDevin, 2017-12-08:
Oh yes...  It should wait a little longer for the terminating LF to arrive, too:

```
    if (gps.available() && (c == 10)) {
```

Otherwise, it would update the LCD while the CR/LF is coming in, and the tablet would see a longer pause before the last sentence was sent.

thomaszietz, 2017-12-09:
Hi,
perfect - I'll test it in a few days.
Thanks a lot!
Thomas

closed 2017-12-24 by SlashDevin