SlashDevin/NeoGPS #66: Not getting STM32 to work

Gerno082, 2018-01-09:
I recently bought a STM32F103 board and can't get the Arduino IDE to compile for the board.   I am not even sure if the library is compatible with this board(1st error line).  I plan to communicate over Serial2.  I get a whole lot of errors, I have no idea where to begin.  

Arduino: 1.8.4 (Windows 7), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), Serial, 72Mhz (Normal), Smallest (default)"

Invalid library found in C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master: C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master

WARNING: library NeoGPS claims to run on (avr, samd, sam, esp8266, arc32) architecture(s) and may be incompatible with your current board which runs on (STM32F1) architecture(s).

WARNING: library AltSoftSerial-master claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (STM32F1) architecture(s).

In file included from C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:35:0:

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\config/AltSoftSerial_Boards.h:146:2: error: #error "Please define your board timer and pins"

 #error "Please define your board timer and pins"

  ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp: In static member function 'static void AltSoftSerial::init(uint32_t)':

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:74:27: error: 'CONFIG_TIMER_NOPRESCALE' was not declared in this scope

   CONFIG_TIMER_NOPRESCALE();

                           ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:79:28: error: 'CONFIG_TIMER_PRESCALE_8' was not declared in this scope

    CONFIG_TIMER_PRESCALE_8();

                            ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:104:10: error: 'INPUT_CAPTURE_PIN' was not declared in this scope

  pinMode(INPUT_CAPTURE_PIN, INPUT_PULLUP);

          ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:105:15: error: 'OUTPUT_COMPARE_A_PIN' was not declared in this scope

  digitalWrite(OUTPUT_COMPARE_A_PIN, HIGH);

               ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:113:27: error: 'ENABLE_INT_INPUT_CAPTURE' was not declared in this scope

  ENABLE_INT_INPUT_CAPTURE();

                           ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp: In static member function 'static void AltSoftSerial::end()':

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:118:24: error: 'DISABLE_INT_COMPARE_B' was not declared in this scope

  DISABLE_INT_COMPARE_B();

                        ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:119:28: error: 'DISABLE_INT_INPUT_CAPTURE' was not declared in this scope

  DISABLE_INT_INPUT_CAPTURE();

                            ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:122:24: error: 'DISABLE_INT_COMPARE_A' was not declared in this scope

  DISABLE_INT_COMPARE_A();

                        ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp: In static member function 'static void AltSoftSerial::writeByte(uint8_t)':

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:138:15: error: 'SREG' was not declared in this scope

  intr_state = SREG;

               ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:139:6: error: 'cli' was not declared in this scope

  cli();

      ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:147:24: error: 'ENABLE_INT_COMPARE_A' was not declared in this scope

   ENABLE_INT_COMPARE_A();

                        ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:148:22: error: 'CONFIG_MATCH_CLEAR' was not declared in this scope

   CONFIG_MATCH_CLEAR();

                      ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:149:33: error: 'GET_TIMER_COUNT' was not declared in this scope

   SET_COMPARE_A(GET_TIMER_COUNT() + 16);

                                 ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:149:39: error: 'SET_COMPARE_A' was not declared in this scope

   SET_COMPARE_A(GET_TIMER_COUNT() + 16);

                                       ^

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp: At global scope:

C:\Users\Johannes\Documents\Arduino\libraries\AltSoftSerial-master\AltSoftSerial.cpp:155:4: error: expected constructor, destructor, or type conversion before '(' token

 ISR(COMPARE_A_INTERRUPT)

    ^

exit status 1
Error compiling for board Generic STM32F103C series.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

SlashDevin, 2018-01-09:
These errors are from trying to use AltSoftSerial, a software alternative to HardwareSerial ports.  Since you want to use `Serial2`, follow the [instructions for changing GPSport.h](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Installing.md#4-review-librariesneogpssrcgpsporth).

The first warning is that the NeoGPS library.properties file does not yet list STM32, although I have reports that it does work.  It's only a warning, so try getting your GPSport.h file fixed first.

Gerno082, 2018-01-09:
Thanks for the advice, I believe I solved the problem with the GPSport.h, but still receive the following: 

C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master\src\NMEAGPS.cpp: In member function 'NMEAGPS::decode_t NMEAGPS::parseCommand(const NMEAGPS::msg_table_t*, uint8_t, char)':

C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master\src\NMEAGPS.cpp:566:84: error: 'pgm_read_ptr' was not declared in this scope

       const char *        table_i = (const char *        ) pgm_read_ptr( &table[i] );

                                                                                    ^

C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master\src\NMEAGPS.cpp: In member function 'const __FlashStringHelper* NMEAGPS::string_for(NMEAGPS::nmea_msg_t) const':

C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master\src\NMEAGPS.cpp:651:76: error: 'pgm_read_ptr' was not declared in this scope

       const char *        table_i = (const char *) pgm_read_ptr( &table[i] );

                                                                            ^

C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master\src\NMEAGPS.cpp: In static member function 'static void NMEAGPS::poll(Stream*, NMEAGPS::nmea_msg_t)':

C:\Users\Johannes\Documents\Arduino\libraries\NeoGPS-master\src\NMEAGPS.cpp:1997:82: error: 'pgm_read_ptr' was not declared in this scope

       (const __FlashStringHelper *) pgm_read_ptr( &poll_msgs[msg-NMEA_FIRST_MSG] );

                                                                                  ^

exit status 1
Error compiling for board Generic STM32F103C series.

SlashDevin, 2018-01-10:
Yes, that is a bug in the STM core.  You can modify NeoGPS_cfg.h to provide the correct definition on line 104.  Simply comment out the `#if` and `#endif` lines:

```
// #if defined(ARDUINO_SAMD_MKRZERO) | defined(ARDUINO_SAMD_ZERO) |   \
    (defined(TEENSYDUINO) && (TEENSYDUINO < 139))
  #undef pgm_read_ptr
  #define pgm_read_ptr(addr) (*(const void **)(addr))
// #endif
```

Gerno082, 2018-01-10:
Thanks for all the help and advice, I uploaded a few sketches and it worked perfectly.

Regards
Gerno

closed 2018-01-11 by SlashDevin