SlashDevin/NeoGPS #67: Satellites view example questions

thomasnisole, 2018-01-17:
Hey,

I'm using your library to make a sailing GPS with a TFT screen, and i want to display a satellites view even when the GPS is not connected. 
I saw your example but i don't understand how i can make it.

So, how update your example to take GSV sentences even GPS module not connected ? Is it possible to help me ?

Thanks in advance.

SlashDevin, 2018-01-17:
> i want to display a satellites view even when the GPS is not connected.

I don't understand your question.

The GPS device emits GSV sentences.  NeoGPS parses those characters and fills out the array `gps.satellites[]`.

If the GPS device is not connected,
&nbsp;&nbsp;&nbsp;&nbsp;==> no sentences would be received by the Arduino,
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;==>  NeoGPS (in your sketch) would not parse any characters
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;==> `gps.satellites[]` would always be empty

Do you want to display an EMPTY satellite view on your TFT when the GPS device is not connected?

>  how update your example to take GSV sentences even GPS module not connected ?

What is the *source* of the GSV sentences when the GPS device is not connected?

thomasnisole, 2018-01-17:
I make a mistake in my explanations. The GPS module is connected physically to the arduino, but i want to receive satellites informations even when the GPS can not yet give any coordinates. 

Is it more clearly ?

Thanks.

SlashDevin, 2018-01-17:
>  i want to receive satellites informations even when the GPS can not yet give any coordinates.

The [NMEAGSV](https://github.com/SlashDevin/NeoGPS/blob/master/examples/NMEAGSV/NMEAGSV.ino) example displays all satellites that the GPS device is receiving, even if the location is not valid (i.e.`!valid.location`).  It also displays satellites that are not (or cannot) be used to calculate the current position.  This is indicated by the `tracked` flag.

Without seeing your sketch, I can only guess that you are testing `valid.location` and not displaying the satellites that may be in the `gps.satellites[]` array.

There are other configuration mistakes that could cause trouble, too.  Choosing the correct LAST_SENTENCE is crucial, so **be sure to run NMEAorder.ino**.

thomasnisole, 2018-01-18:
Hey @SlashDevin,

Thanks for your help.
Here is my code (your code, it's your example) with my configuration (i test on an arduino Mega 2560 with NEO-7M Gps module) :

```cpp
#define RX_PIN 51
#undef SERIAL_PORT_HARDWARE_OPEN

#define NMEAGPS_PARSE_GSV 1
#define NMEAGPS_PARSE_SATELLITES 1
#define NMEAGPS_PARSE_SATELLITE_INFO 1
#include <NeoSWSerial.h>

#include <NMEAGPS.h>

NMEAGPS  gps; // This parses the GPS characters
gps_fix  fix; // This holds on to the latest values

#include <GPSport.h>
#define LAST_SENTENCE_IN_INTERVAL NMEAGPS::NMEA_GLL

void setup()
{
  DEBUG_PORT.begin(9600);
  while (!Serial)
    ;
  DEBUG_PORT.print( F("NeoGPS GSV example started\n") );

  gpsPort.begin(9600);

} // setup

//-----------------

void loop()
{
  while (gps.available( gpsPort )) {   
    fix = gps.read();
    displaySatellitesInView();  
  }
} // loop

//-----------------

void displaySatellitesInView()
{
  DEBUG_PORT.print( gps.sat_count );
  DEBUG_PORT.print( ',' );

  for (uint8_t i=0; i < 20; i++) {
    DEBUG_PORT.print( gps.satellites[i].id );
    DEBUG_PORT.print( ' ' );
    DEBUG_PORT.print( gps.satellites[i].elevation );
    DEBUG_PORT.print( '/' );
    DEBUG_PORT.print( gps.satellites[i].azimuth );
    DEBUG_PORT.print( '@' );
    if (gps.satellites[i].tracked)
      DEBUG_PORT.print( gps.satellites[i].snr );
    else
      DEBUG_PORT.print( '-' );
    DEBUG_PORT.print( F(", ") );
  }

  DEBUG_PORT.println();

} // displaySatellitesInView
```

With same configuration (RX on 51with NeoSWSerial), i was able to run NMEALoc with success.

I put few debug trace in the previous code, and i saw the sketch does'nt pass the test `gps.available(gpsPort)`, or in NMEALoc example it pass.

I don't understand why...

Thanks for your help.

SlashDevin, 2018-01-18:
You can't put configuration items in your sketch.  You must modify the configuration files in the `Libraries/NeoGPS/src` directory.  Bad things happen.

thomasnisole, 2018-01-18:
OK, but if another developer must work on the project he have to update the config in the library, it's not good practice, a library must provide a system to accept an external configuration. 

About Serial1, 2 and 3, they are lock by the screen shield.

So, thank you for your time and your help.

SlashDevin, 2018-01-18:
> a library must provide a system to accept an external configuration.

It's actually the Arduino IDE that prevents local includes.  I've complained about this many times, but the Arduino gods just aren't going to change their build process.

You may be able to copy the entire NeoGPS/src directory into your sketch directory to make the entire thing a "local" version.  Then edit the config files locally, and use the double-quoted includes in your sketch, instead of angle brackets:

`#include "NMEAGPS.h"`

But there's no way to have the config files in the sketch directory and everything else in a shared directory.  :(

If you don't do one of these 2 things (edit the src config files in Libraries/NeoGPS or in the local copy in the sketch directory), the src/.cpp files get built with the src config, and your sketch gets built with the #define configs.  Then the sketch object file does not match the NeoGPS object files, and Bad Things Happen (struct sizes are different, etc.).

> About Serial1, 2 and 3, they are lock by the screen shield.

Really?  Shields usually don't overhang pins 18 & 19.  I would even suggest removing the socket header from pins 14-21 and soldering wires to those pins for the GPS, or perhaps a right-angle socket header.  Anything but a software serial solution.  :P  You might be able to jam a wire in there before seating the shield.

closed 2018-01-23 by SlashDevin