SlashDevin/NeoGPS #68: NeoGPS and AT modems with GPS

dmnc-net, 2018-01-24:
Hi, I'm using SIM808 shield with my Uno (`NeoSWSerial` between them).

I'm reading the examples and src files and I'm trying to find out, where and how (efficiently) to put AT commands to the sketch because without AT commands, my GPS do nothing. Is the NeoGPS even suitable for this application?

My goal is to replace @vshymanskyy's TinyGPS (I'm considering it as very dirty code ... sorry Volodymyr, i.e. lot of `String` objects and `int` in places where `uint8_t` is enough) to my own solution with NeoGPS as a parser.

SlashDevin, 2018-01-25:
I don't know what you're trying to do.

If you are sending an AT command that causes GPS characters (NMEA sentences likle $GPGGA) to start coming out of the SIM808, NeoGPS can parse that stream.

If you are sending an AT command the causes the SIM808 to send one line of GPS information (comma-separated values, not NMEA sentences), you don't need NeoGPS to parse those fields.  `strtok` with `atof` and `atoi`/`atol` would be sufficient.

If there is a separate GPS device from the SIM shield, NeoGPS can parse the NMEA stream, and your sketch can send/receive characters to/from the SIM shield.

vshymanskyy, 2018-01-25:
@dmnc-net Please don't mark people in topics they aren't involved in.
If you see ways to improve OSS projects, Pull Requests are welcome as usual ;)

dmnc-net, 2018-01-26:
@SlashDevin I'm still reading sources of your library and other libraries to understand how everything works and how I can achieve my goal.

I have some shield with SimCom SIM808 (integrated GSM, GPRS, GPS). Without AT commands id does nothing. To get GPS data, I have to write `AT+CGNSINF` to the serial port and receive one line answer. Then repeat. This line contains comma-separated values and I thought it was NMEA because manufacturer's application notes mentioned it (and their English is even worse than mine).

I see, I've found some example and probably this is NMEA (RMC): `$GPRMC,123519,A,4807.038,N,01131.000,E,022.4,084.4,230394,003.1,W*6A`
but this line from my device is already parsed RMC sentence: `1,1,20180126072452.000,40.123456,10.123456,216.000,2.00,354.9,1,,2.8,3.0,1.0,,12,4,,,36,,` (lat/lon redacted due to privacy) ... Am I right? So your response is probably already the answer to my question and hopefully will help to any other lost people like me in the future :)
Thank you.

@vshymanskyy sorry for disturbing you, sir. I've mentioned you in good faith because I came for help to the OSS community which doesn't necessarily have to have sharp boundaries where one is not interested anymore. Eventually it turned out that my goal is more related to your library than NeoGPS.  Thanks for the PRs offer.

closed 2018-01-26 by dmnc-net