SlashDevin/NeoGPS #71: fix for NMEAGPS.cpp and a questuin to get satellite SNR

gitgregsi, 2018-02-27:
Hi, thanks for great code. I discovered an error in NMEAGPS.cpp when compiler did not want to compile after I enabled lat and long error fix ... m_fix.valid.alt_err should be m_fix.valid.lat_err I guess .. at least this fixed the error :) 

bool NMEAGPS::parse_lat_err( char chr )
{
  #ifdef GPS_FIX_LAT_ERR
    if (chrCount == 0)
      NMEAGPS_INVALIDATE( lat_err );
    if (parseFloat( m_fix.lat_err_cm, chr, 2 )) {
      if (validateFields() &&
          (negative || (m_fix.valid.alt_err > MAX_ERROR_CM))) **//should be .lat**
        sentenceInvalid();
      else
        m_fix.valid.lat_err = (chrCount != 0);
    }
  #endif

MY question - I enabled GSV parsing, what function do I use to parse out SNR signal strength for each satellite or at least see it?     trace_all( Serial, gps, fix ); does not print it for example...

thx!

gitgregsi, 2018-02-27:
Got the SNR working so no need for answer on that part :)

SlashDevin, 2018-02-27:
Nice catch!  That's been there for a while...

I hope you found the SNR in `gps.satellites[i].snr`...

SlashDevin, 2018-03-05:
Sorry, I haven't applied a patch for this yet...

gitgregsi, 2018-03-05:
no problem for me, I discovered that my NEO6 GPS isn't putting that info out anyway :))
But the SNR works great!
Thanks

SlashDevin, 2018-03-07:
...there we go.  :)

commit 4930adbc (2018-03-07): Fix #71

* Copy & Paste error
closed 2018-03-07 by SlashDevin