SlashDevin/NeoGPS #74: GPS port error

Speculsion, 2018-03-12:
Whenever I use 'gps_port' the program comes up with this error.
Code: [https://gist.github.com/Speculsion/6ec0c0c52696c48c2129e76fad1bf8b7](url)
Error:
`./opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/051687342/build -hardware opt/arduino-builder/hardware -hardware ./opt/cores -tools opt/arduino-builder/tools -tools ./opt/tools -built-in-libraries opt/libraries/latest -libraries /tmp/051687342/pinned -libraries /tmp/051687342/custom -fqbn arduino:avr:uno -build-cache /tmp -logger humantags -verbose=false /tmp/051687342/V2HighAltitudeBalloonDataLogger

Multiple libraries were found for "SoftwareSerial.h"

Used: /home/admin/builder/opt/cores/arduino/avr/libraries/SoftwareSerial

Not used: /home/admin/builder/opt/libraries/latest/printoo_library-1-0-2

Multiple libraries were found for "AltSoftSerial.h"

Used: /home/admin/builder/opt/libraries/latest/altsoftserial-1-4-0

Not used: /home/admin/builder/opt/libraries/latest/cmmc_nb-iot-0-0-2

Not used: /home/admin/builder/opt/libraries/latest/barebonesim800-1-1-0

Not used: /home/admin/builder/opt/libraries/latest/ais_nb_bc95-1-0-4

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino: In function 'void sendUBX(uint8_t*, uint8_t)':

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino:77:5: error: 'gps_port' was not declared in this scope

gps_port.write(MSG[i]);

^

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino:80:3: error: 'gps_port' was not declared in this scope

gps_port.println();

^

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino: In function 'boolean getUBX_ACK(uint8_t*)':

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino:124:9: error: 'gps_port' was not declared in this scope

if (gps_port.available()) {

^

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino: In function 'void GPSloop()':

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino:252:25: error: 'gps_port' was not declared in this scope

while (gps.available( gps_port ))

^

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino: In function 'void setup()':

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino:266:3: error: 'gps_port' was not declared in this scope

gps_port.begin(9600);

^

/tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino:287:167: error: expected ')' before 'USING_GPS_PORT'

Serial.println( F("Looking for GPS device on " USING_GPS_PORT) );

^

exit status 1`

Speculsion, 2018-03-12:
Sorry here is the updated link:
 https://gist.github.com/Speculsion/6ec0c0c52696c48c2129e76fad1bf8b7

SlashDevin, 2018-03-13:
I think you might have an INO file that uses the old GPS port name.  Replace `gps_port` with `gpsPort`.

Speculsion, 2018-03-13:
Thanks this really helped a lot but still won't fix this error:
/
tmp/051687342/V2HighAltitudeBalloonDataLogger/V2HighAltitudeBalloonDataLogger.ino:287:167: error: expected ')' before 'USING_GPS_PORT'

Serial.println( F("Looking for GPS device on " USING_GPS_PORT) );

^

exit status 1`

Thanks for solving the bulk of this issue! :)

SlashDevin, 2018-03-13:
That preprocessor symbol has also changed.  All the example programs now use `GPS_PORT_NAME` instead of `USING_GPS_PORT`.   Full description of the symbols expected are [here](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Installing.md#choosing-your-own-serial-port)

Speculsion, 2018-03-14:
Thanks! This issue has been resolved.

closed 2018-03-14 by SlashDevin