SlashDevin/NeoGPS #76: Error compiling for board Arduino Due

overexposer, 2018-03-19:
With versions 4.2.4 and 4.2.5 I get this compiling error.

In file included from C:\Users\Pmac\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/Arduino.h:31:0,

                 from C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.h:23,

                 from C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp:18:

C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp: In member function 'NMEAGPS::decode_t NMEAGPS::parseCommand(const NMEAGPS::msg_table_t*, uint8_t, char)':

C:\Users\Pmac\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type

 #define pgm_read_ptr(addr) (*(const void *)(addr))

                                                 ^

C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp:566:60: note: in expansion of macro 'pgm_read_ptr'

       const char *        table_i = (const char *        ) pgm_read_ptr( &table[i] );

                                                            ^

C:\Users\Pmac\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type

 #define pgm_read_ptr(addr) (*(const void *)(addr))

                                                 ^

C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp:590:49: note: in expansion of macro 'pgm_read_ptr'

         const char *table_next = (const char *) pgm_read_ptr( &table[next_msg] );

                                                 ^

C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp: In member function 'const __FlashStringHelper* NMEAGPS::string_for(NMEAGPS::nmea_msg_t) const':

C:\Users\Pmac\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type

 #define pgm_read_ptr(addr) (*(const void *)(addr))

                                                 ^

C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp:651:52: note: in expansion of macro 'pgm_read_ptr'

       const char *        table_i = (const char *) pgm_read_ptr( &table[i] );

                                                    ^

C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp: In static member function 'static void NMEAGPS::poll(Stream*, NMEAGPS::nmea_msg_t)':

C:\Users\Pmac\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:106:49: error: 'const void*' is not a pointer-to-object type

 #define pgm_read_ptr(addr) (*(const void *)(addr))

                                                 ^

C:\Users\Pmac\Documents\Arduino\libraries\NeoGPS\src\NMEAGPS.cpp:1997:37: note: in expansion of macro 'pgm_read_ptr'

       (const __FlashStringHelper *) pgm_read_ptr( &poll_msgs[msg-NMEA_FIRST_MSG] );

                                     ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

I receive the same error using any of the provided examples. I've even opened a new sketch and #include <NEMAGPS.h> by its self and get the same error. Version 4.2.3 and older are the only stable versions that compile with no issues.

SlashDevin, 2018-03-19:
This is actually an error in the Due core.  :-/  I have added similar work-arounds for other platforms.  I'll check in a new release shortly...

closed 2018-03-19 by SlashDevin

SlashDevin, 2018-04-08:
Thanks, I'll add it to the next update.

commit 1d18418e (2018-03-19): Fix #76

Work-around for bad `pgm_read_ptr` in Due core.