SlashDevin/NeoGPS #79: SyncTime: class NMEAGPS' has no member named 'UTCms'

al-ang, 2018-03-31:
I'm looking for a small footprint library to get the time from my neo6m gps on a nano v3 and tried to compile SyncTime to see what it might provide.  Got the compile error listed below.  Please close issue if this a dumb thing on my part, but I thought I'd mention it in case it was a real issue:

GPS_SyncTime:136: error: 'class NMEAGPS' has no member named 'UTCms'
     uint32_t UTCms = gps.UTCms();
exit status 1
'class NMEAGPS' has no member named 'UTCms'

SlashDevin, 2018-04-01:
To make the UTCms() function available, you have to define one of the TIMESTAMP configuration options in NMEAGPS_cfg.h.  I get a compile error with the original SyncTime.ino example if one of them is not defined:

```
SyncTime:58: error: #error You must define NMEAGPS_TIMESTAMP_FROM_INTERVAL or PPS in NMEAGPS_cfg.h!

   #error You must define NMEAGPS_TIMESTAMP_FROM_INTERVAL or PPS in NMEAGPS_cfg.h!

    ^
```
It looks like you are using your own sketch, so it probably does not have this `#error` message to help you.

al-ang, 2018-04-01:
Thank you!

From: SlashDevin 
Sent: Saturday, March 31, 2018 6:09 PM
To: SlashDevin/NeoGPS 
Cc: al-ang ; Author 
Subject: Re: [SlashDevin/NeoGPS] SyncTime: class NMEAGPS' has no member named 'UTCms' (#79)

To make the UTCms() function available, you have to define one of the TIMESTAMP configuration options in NMEAGPS_cfg.h. I get a compile error with the original SyncTime.ino example if one of them is not defined:

SyncTime:58: error: #error You must define NMEAGPS_TIMESTAMP_FROM_INTERVAL or PPS in NMEAGPS_cfg.h!

   #error You must define NMEAGPS_TIMESTAMP_FROM_INTERVAL or PPS in NMEAGPS_cfg.h!

    ^
It looks like you are using your own sketch, so it probably does not have this #error message to help you.

closed 2018-04-01 by SlashDevin