SlashDevin/NeoGPS #81: Issue with gps.satellies[i]

mabbotts9797, 2018-04-07:
[According to the docs](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Data%20Model.md) you may access more detailed information regarding satellites with the gps.satellies[] array. I have tried to loop through the elements of the array but this fails with:

`class NMEAGPS' has no member named 'satellies'`

I also tried **gps.satellites** but this also fails.

Is there a specific way I am supposed to be going about this?

SlashDevin, 2018-04-07:
You must enable these members with several configuration items.  In NMEAGPS_cfg.h:

```
#define NMEAGPS_PARSE_GSV

#define NMEAGPS_PARSE_SATELLITES
#define NMEAGPS_PARSE_SATELLITE_INFO
```

The GSV sentences provide this information, so they must be parsed.  The two SATELLITES defines enable the satellites array in the `gps` object.

You must also configure the GPS device to _send_ the GSV sentences.  Different brands have different commands that you send to them.

Try the NMEAGSV.ino example program for the exact techniques.  It will not build until you have the configuration correct.

mabbotts9797, 2018-04-07:
Thanks! Managed to get it working. Much appreciated. As for the satellite ID's... How do I go about looking those up? Is the ID a standard identifier in some sort of lookup system?

SlashDevin, 2018-04-07:
There is no "standard" ID that applies to all Global Navigation Satellite Systems (GNSS's).  Each system has its own list of satellite IDs (start [here](https://en.wikipedia.org/wiki/Satellite_navigation)).

However, GPS device manufacturers have their own way of _reporting_ which satellites are being used.  You would have to look at the device spec to see if (1) your device supports multiple GNSS's, and (2) how satellites are numbered in the GSV sentences.

Complicating the issue is the NMEA "Talker ID".  Most applications expect to receive sentences like GPRMC, GPGGA or GPGSV.  The first two characters actually identify which GNSS is "talking".  For those sentences, **GP**S is the talker.  The device could have sent **GL**RMC for information derived from the GLONASS satellites.  It may even send **GN**RMC, which means "multiple GNSS's" were used to produce the RMC information.  There are ways to distinguish this in NeoGPS.

If your device can produce GNGSV sentences, it must have a way of numbering the satellites distinctly.  This is specified by the device manufacturer.

Which GPS device are you using?

mabbotts9797, 2018-04-07:
Thanks for the information. I guess it was naive to assume there would be some common standard. I am using the **ublox NEO-6M-0-001** GPS module. I have just downloaded the datasheet so hopefully, it can shed some light on the issue at hand.

mabbotts9797, 2018-04-07:
Page 60 on  [this datasheet](https://www.u-blox.com/sites/default/files/products/documents/u-blox6_ReceiverDescrProtSpec_%28GPS.G6-SW-10018%29_Public.pdf) states the identifier used is SV(N). The exact quote: 

> The SV Numbers (Fields 'Sv') are in the range of 1 to 32 for GPS satellites, and 33 to 64
for SBAS satellites (33 = SBAS PRN 120, 34 = SBAS PRN 121, and so on)

I think this means I can look up the IDs using the SVN column [here.](https://en.wikipedia.org/wiki/List_of_GPS_satellites)

closed 2018-04-07 by mabbotts9797