SlashDevin/NeoGPS #85: Reading an AT Command Response on Serial or from a char array

abrileroux, 2018-04-29:
Hi there,

Firstly I apologize for asking this question using this medium. I'm busy with a project using a Seeed Wio Tracker. The unit has a Quectel M20 GPS/GPRS module which talks to the Atmel microprocessor via Serial1. 

I have to query the module via the AT Command "AT+QGNSSRD?\n\r", which will then once (and only once, until the AT command is called again), respond with on Serial1. 

> AT+QGNSSRD?
> 
> +QGNSSRD: $GNRMC,200101.082,A,3351.4838,S,01830.6949,E,0.54,6.38,290418,,,A*64
> $GNVTG,6.38,T,,M,0.54,N,0.99,K,A*2F
> $GNGGA,200101.082,3351.4838,S,01830.6949,E,1,7,1.30,21.7,M,32.6,M,,*6A
> $GPGSA,A,3,23,22,29,03,16,25,31,,,,,,1.59,1.30,0.91*0E
> $BDGSA,A,3,,,,,,,,,,,,,1.59,1.30,0.91*14
> $GPGSV,3,1,09,26,88,289,,16,56,323,29,31,52,137,32,33,36,309,*76
> $GPGSV,3,2,09,22,35,292,31,03,34,259,36,23,23,225,34,29,21,122,29*73
> $GPGSV,3,3,09,25,05,137,30*44
> $BDGSV,1,1,00*68
> $GNGLL,3351.4838,S,01830.6949,E,200101.082,A,A*59
> 
> OK

I have managed to read this whole statement (including the AT echo and the OK), into a char array buffer. 

It would mean a lot to me if someone could please point me in the right direction as to how one would get this data into NeoGPS? 

Thank you so much

SlashDevin, 2018-04-30:
First, make sure that LAST_SENTENCE is set to GLL in NMEAGPS_cfg.h.

Then you can feed the entire array to NeoGPS:

```
  char *ptr = &buffer[0];
  while (*ptr)
    gps.handle( *ptr++ );

  if (gps.available()) {
    fix = gps.read();
       ...
```

NeoGPS will ignore everything but the NMEA sentences.  They all start with a $ character.

BTW, you don't really have to buffer all of this... you could feed the response characters to NeoGPS as they arrive.  Just test for a fix becoming available:

```
void loop()
{
  if (timeToGetFix()) {
    M20.println( F( "AT+QGNSSRD?\n\r") );
  }
        ....

  if (M20.available()) {
    char c = M20.read();

    // Handle command responses
       ... watch for \r\nOK\r\n ? ...

    // Handle GPS data
    gps.handle( c );
    if (gps.available()) {
      fix = gps.read();
         ...
    }
  }
```

abrileroux, 2018-04-30:
Thank you so much!

I really appreciate your help.

Keep up the good work!

closed 2018-04-30 by abrileroux