SlashDevin/NeoGPS #86: Class/Example for u-blox GNSS attached to Wire/I2C.

mazgch, 2018-05-02:
I wrote a simple class that can be used in combination with your NMEAGPS library. The class allows you to use the Wire interface in combination with a u-blox GNSS device attached to Wire/I2C as a direct source for the Stream. This works for me and I thought you might like to be aware or even add functionality to your class as part of the library or an example. 

[GpsTwoWire.txt](https://github.com/SlashDevin/NeoGPS/files/1967094/GpsTwoWire.txt)

afnan, 2020-01-04:
@mazgch  i was trying to use your code but could not. Do you have a library or instructions on how to use this code?

mazgch, 2020-01-04:
Sparkfun has now their own library that supports i2c https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library