SlashDevin/NeoGPS #87: How to enable Lat/Lon Error

wezside, 2018-05-03:
I'm trying to use lat_err() and lon_err() for a simplified Kalman Filter but struggling to enable this for my GPS device. It's a Ublox Neo-6M module. Apart from GPSfix_cfg.h where I've uncommented both GPS_FIX_LAT_ERR and GPS_FIX_LON_ERR what else should be enabled? 

Based on issue #39 it would appear I need to enable NMEAGPS_PARSE_GST. But this doesn't work either. I'm a bit of newbie here but this is by far the best GPS lib I've used.

SlashDevin, 2018-05-03:
Yes, you must enable those fields in GPSfix_cfg.h, and you must enable a sentence that would set those fields in NMEAGPS_cfg.h (i.e. enable the GST sentence parsing).  You must also make sure your GPS device is actually _sending_ a GST sentence.  You can send a PUBX,40 configuration command in `setup` to do that:

```
  gps.send_P( &gpsPort, F("PUBX,40,GST,0,1,0,0,0,0") ); // enable GST sentence, one per update interval
```
See the ublox NEO-6 Protocol spec for more information.

wezside, 2018-05-03:
Thank you. I thought it might be hardware related. Didn't know about the PUBX command so will read a bit more about the spec.

wezside, 2018-05-04:
After looking at the U-blox 6M protocol it appears I need to use the GBS Message which falls under "NMEA Standard Messages" to enable *GNSS Satellite Fault Detection*. Do I need to enable any specific parser for this or will this message be parsed by NEOGPS automatically?

SlashDevin, 2018-05-04:
Both GST and GBS messages report the standard deviation of the lat/lon/alt errors.

What part of your "simplified Kalman filter" (not Madgwick?) would use the "most likely failed satellite ID", "estimate on most likely failed satellite", or "standard deviation of estimate on most likely failed satellite"?  The "probabilty of missed detection" is not supported, and will always be an empty field.  Are you sure you need the extra insight into the NEO-6 RAIM process?  

You could use the GSV sentences to know which satellite IDs are not being used, have a weak signal, or are close to the horizon.  There are also the DOP values that provide a qualitative measure of the current constellation.

I'm going to move the GBS request to new Issue #88, as an "enhancement".  We can continue a discussion here.

wezside, 2018-05-04:
I think I need to come clean in that I'm no expert in GPS protocol. How I got here was simply looking for a way to better smooth the "jumps" in lat/lng positional data. My device I'm building measures speed on water using a GPS receiver. I've done some searching and implemented a median-5 approach but all info indicated that a Kalman Filter would be better. So I found this [QA](https://stackoverflow.com/questions/1134579/smooth-gps-data) (2nd answer) about an implementation where the covariance matrix is not needed, hence the wording Simplified Kalman Filter. It works well for Android because Android's GPS receiver includes error data. So I saw NEOGPS have this feature which brings me to this thread. 

I looked through the 6M protocol document and found **errlat** and **errlng** under the GBS  standard message. If I'm barking up the wrong constellation by all means feel free to correct me.

SlashDevin, 2018-05-05:
> I looked through the 6M protocol document and found errlat and errlng under the GBS standard message. 

Right, they are also in the GST sentence.  NeoGPS does cannot parse a GBS sentence, but it can parse a GST sentence (if enabled and sent by the device). Just follow the steps in my first reply.

wezside, 2018-05-05:
Great thanks for the time.

closed 2018-05-05 by wezside