SlashDevin/NeoGPS #89: Ublox Neo-6M from great to nothing

wolfgangrepository, 2018-05-06:
Hello SlashDevin,
We're a Ublox Neo-6M, and using your NeoGPS package of goodness. It's worked well up til now, and we've unfortunately run into an issue that we are finding difficult to resolve. Hoping you can help us out.

First and foremost, how delicate have you found the Ublox NeoGPS devices to be? The first vital bit of info is that the light on the module no longer turns on, and this tells us it's possible that the module has been fried, though we've taken great care to make sure all wiring is correct. The only changes that have been made since its functioning state is the location... is it possible for static to fry these modules?

Running NMEAloc gives us a single column of question marks. 

Running NMEA we get the following:
_`NMEA.INO: started
  fix object size = 31
  gps object size = 84
Looking for GPS device on AltSoftSerial( RX pin -1, TX pin -1 )

GPS quiet time is assumed to begin after a RMC sentence is received.
  You should confirm this with NMEAorder.ino

Status,UTC Date/Time,Lat,Lon,Hdg,Spd,Alt,Sats,Rx ok,Rx err,Rx chars,
0,,,,,,,,1,0,23,
0,,,,,,,0,7,0,185,
0,,,,,,,0,13,0,347,
0,,,,,,,0,19,0,509,
0,,,,,,,0,25,0,671,
0,,,,,,,0,31,0,833,
0,,,,,,,0,37,0,995,
0,,,,,,,0,43,0,1157,
0,,,,,,,0,49,0,1319,
0,,,,,,,0,55,0,1481,`_

which tells me that its racking up Rx oks and Rx chars, but nothing else.

Running NMEAtest gives us an error message, that essentially reads out the line:
_#error NMEAGPS_PARSE_GGA, GLL, GSA, GSV, RMC, VTG and ZDA must be defined in NMEAGPS_cfg.h!_
We have made this correction in NMEAGPS_cfg.h, but it does not resolve our problem.

It was working perfectly yesterday, hoping this is a resolvable issue. Would greatly appreciate any troubleshooting tips you may have for us. 

Thank you.

Best,
wolfgang.repository

SlashDevin, 2018-05-06:
The GPS device appears to be working:

```
Status,UTC Date/Time,Lat,Lon,Hdg,Spd,Alt,Sats,Rx ok,Rx err,Rx chars,
0,,,,,,,,1,0,23,
0,,,,,,,0,7,0,185,
```
`Rx ok` counts the number of NMEA sentences that were successfully parsed.  Because that number is increasing, I know that the NEO-6M is sending data to the Arduino, and NeoGPS is correctly parsing the characters.

Because the `Status` field is always 0, and the `Sats` field is always 0, I can say that the NEO-6M is not able to receive any satellites.  I would check the antenna connection, if it is connected with a short cable.  Of course, you must have a clear view of the sky.  You can try moving closer to a window, but you may have to go outside.

If the antenna is soldered directly to the board, it is possible that you have damaged the module in some way.  Inspect the solder joints and traces on the board for some kind of mechanical damage.

You should use NMEAorder.ino instead of NMEAtest.ino.  NMEAtest.ino is only for testing the parser -- it does not read from the NEO-6M device.  Revert the configuration files to their original state (RMC & GGA only, etc.).  NMEAorder should tell you to use GLL for the LAST_SENTENCE.

SlashDevin, 2018-05-06:
Sorry, didn't mean to close it yet.   :)

wolfgangrepository, 2018-05-07:
Thanks for such a thorough response!
I’ve undone the changes we made to the library, and running NMEAorder does tell us to use GLL for the LAST_SENTENCE. The more “in-the-know” member of our team is working on this change (and a couple of other ideas he has in the works), will let you know how this goes.

wolfgangrepository, 2018-05-08:
When we receive the message from NMEAorder that says  we need to use GLL for LAST_SENTENCE, what is the actual code that we must must change in the NMEA_cfg.h file? The original file has "#define LAST_SENTENCE_IN_INTERVAL (nmea_msg_t) NMEA_LAST_MSG+5 /* ubloxNMEA::PUBX_04" and I am under the impression that I am to *delete* this line and put in its place "#define LAST_SENTENCE_IN_INTERVAL NMEAGPS::NMEA_GLL". Is this correct? Or am I misunderstanding the error message?

SlashDevin, 2018-05-08:
You did not revert NMEA_cfg.h to [the original version](https://github.com/SlashDevin/NeoGPS/blob/master/src/NMEAGPS_cfg.h#L48).  The original version has this line:

```
#define LAST_SENTENCE_IN_INTERVAL NMEAGPS::NMEA_RMC
```
Change that to a GLL:
```
#define LAST_SENTENCE_IN_INTERVAL NMEAGPS::NMEA_GLL
```
You are correct that you should delete the "...+5" definition, but I'm concerned that you have not reverted the config files to their original versions.  I can't guess what else might be different.  Has your "in-the-know" person changed other source files?

wolfgangrepository, 2018-05-09:
Perfect. Thank you! I was able to locate those lines of code and have changed all RMC's to GLL. 

Another question I had for you was about storing the latitude and longitude values from NMEAloc.h and using them to do some work. We are wanting to use these values for our motor control but are running into some error. If I am not mistaken, the lat/long values are acquired inside the `static void doSomeWork( const gps_fix & fix );` function? Since they are declared here, that mean's that we cannot access these values inside of other functions, correct? For example, let's say we obtain a latitude reading from the GPS and we want to compare it to the previous reading and then use the delta of those two readings to determine how much our motor turns. Let's say the motor code is in it's own function `void cylinderGNC()`, how can we get the latitude values out of `static void doSomeWork( const gps_fix & fix );` and into  `void cylinderGNC()` ? Or if we just calculate the difference of the readings inside doSomeWork, how could I use that value inside of cylinderGNC ? Sorry if that is asking too much. Any help would be greatly appreciated!!

SlashDevin, 2018-05-10:
> Since [the fix is] declared here, that mean's that we cannot access these values inside of other functions

Sure, just make it a global:

```
   ...
NMEAGPS gps;
gps_fix fix; // global structure for any part of your sketch to access
   ...
void loop()
{
  if (gps.available( gpsPort )) {
    fix = gps.read(); // assign to global
    doSomeWork( fix );
  }
}
```
The example sketch NMEA.ino does it this way.

wolfgangrepository, 2018-05-13:
Great, everything is working as hoped! Thank you for all of your help here as we stumbled through this issue. Top notch package you have here, and great responsiveness!

closed 2018-05-13 by SlashDevin

wolfgangrepository, 2018-05-17:
Trying to follow the comments in the cfg files and GPSport but am struggling to switch from AltSoftSerial to HardwareSerial. What all needs to be uncommented/changed in the library to run HardwareSerial?

SlashDevin, 2018-05-17:
You only have to change GPSport.h.  Did you read [this](https://github.com/SlashDevin/NeoGPS/blob/master/extras/doc/Installing.md#choosing-your-own-serial-port)?

This makes all the NeoGPS examples use that port.  No other library files use GPSport.h, just the example INO files.

Otherwise, you can do your own declarations in your sketch, without including GPSport.h