SlashDevin/NeoGPS #91: Altitude--MSL or wrt ellipsoid?

CF20852, 2018-05-14:
First, let me thank you (SlashDevin) for this library.  I'm adapting it for a project to measure hike elevation gain using various configurations of Microduinos and Arduinos with GPS shields and barometric pressure sensors (e.g., BMP180/280 devices).  I find GPS altitude is usually noisy enough to cause apparent elevation gain and loss where there is none, and Google Maps elevation data (SRTM, I think) does not have good enough resolution to accurately reflect actual elevation when one is hiking on a steep slope.  Moreover, MapMyHike was providing radically different elevation gain data than the elevation profiles generated by Google Earth and Google Maps, so I wanted to try to figure out which was more accurate.  (I have to correct the barometric altimetry data for the local atmospheric pressure, which I can get from NOAA's Aviation Weather Center for a nearby airport.)

When I look at your Data Model document, I see the following:
"an altitude (above ellipsoid, not Mean Sea Level), accessed with"

But when I look at the code in NMEAGPS.cpp, I see it is pulling altitude data out of field 9 of the GxGGA message, which according to the u-blox NEO-6M Receiver Description and Protocol Specification, is MSL altitude.  Can you explain, please?

SlashDevin, 2018-05-14:
LOL, I just noticed that last week.  I'll have to investigate...

SlashDevin, 2018-05-25:
After some research, I see that all GPS devices output altitude WRT the ellipsoid, aka MSL.  This is a reference surface that is a good approximation of the sea level surface, especially at the coasts.

A different reference surface, the "geoid", describes the land surface.  To get your height above ground, you would have to subtract the `fix.geoidHeight()`  from the `fix.altitude()` (these functions return meters).

closed 2018-05-25 by SlashDevin